OrcaSlicer/OrcaSlicer #5706: Extra Perimeters on overhangs disables briding

tome9111991, 2024-06-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1 Beta

### Operating System (OS)

Windows

### OS Version

10

### Printer

Voron 2.4

### How to reproduce

Open the Project and slice.

on layer 177 you can see overhangs instead of bridges
[image]

### Actual results

Too many overhangs where bridges would actually make more sense.

### Expected results

Extra Perimeters on overhangs disabled to many bridges.
it should look like this:
[image]

### Project file & Debug log uploads

[project.zip](https://github.com/user-attachments/files/15841082/project.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tome9111991, 2024-06-17:
Same issue with Prusaslicer.

But Extra perimeter on overhangs with latest Superslicer version works well.

vovodroid, 2024-07-01:
Actuall SS has this feature only for Classic generator. Probably it was somehow broken when extending it to Arachne. Simplest solution is to split code and make it work correct in Classic for now.

Did you open bug in for Prusa?

tome9111991, 2024-07-01:
thanks for the hint you are right only work correct with Classic  in SS.

No i'dont open bug in Prusa maybe i should

closed 2024-10-07 by github-actions[bot]