OrcaSlicer/OrcaSlicer #5709: Add Adaptive layer height min/max setting

LordBoos, 2024-06-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When setting adaptive layer height, the min/max is loaded from printer extruder settings. This is a bit weird to configure all the time. Most of the time I want adaptive layer height to have max layer height at 0.2mm, but default extruder setting is 0.28mm. I don't want to save new maximum for whole extruder.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be great if I could set min/max layer height of the adaptive layer height right in its tool window: 
[image]
Upper and lower bounds of sliders should be taken from extruder min/max setting.

vovodroid, 2024-06-17:
I added such thing to Prusa, code is available in my fork https://github.com/vovodroid/PrusaSlicer

Savvy-Cyber, 2024-07-18:
Found this post by searching for a solution to the same issue, that is for 'adaptive layers' to not go past the 'max layer height' I have set in Orca. So, here's a vote for "yes please" from me!

LordBoos, 2024-10-17:
Still relevant

closed 2025-01-24 by github-actions[bot]

Myoporum, 2025-02-27:
A shame this is still not being considered, I never want .28mm layers when using a 0.4mm nozzle

Glitchy-Tozier, 2026-10-25:
Is there a way to reopen this so it can move forward towards implementation?