OrcaSlicer/OrcaSlicer #5711: Add PA (Pressure Advance) Setting Based on Line Type (Inner Wall, Outter Wall, Top Surface Infill)

markleaf131313, 2024-06-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Effectively PA needs to be tuned for a specific speed and acceleration. For a lot of profiles the speed and accelerations for the different line types are different. Outer walls are slow, Inner walls are fast, and Top Surface infill is the slowest. This leads to even with a tuned PA looking at a printed model would lead to (assuming the PA is tuned for the outer wall):

* Good sharp corners with no gap for the outside of the model, PA is tuned.
* Over extrusion for the Top Surface, as it is going slower than the outer wall speeds it leads to extra material being left behind, PA is too low.
* For the inner wall that are printed faster there are visible gaps in the corners on the top surface of the (usually 2) inner walls, PA is too high. 

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Adding additional options for PA (pressure advance) to be able to set the pressure advance either by speed, or by line types that are divided in the same way as the speed settings. So that the appropriate PA setting can be used when appropriate. Not sure what is best, maybe some testing would be needed if scaling linearly would be sufficient. As the speeds set for the line types may not be the speeds that it ends up actually running it.

### Describe alternatives you've considered

Setting speeds to be the same so that the PA is what it should be for all line types. This isn't that realistic but for outer walls and top surfaces they are similar enough to be set to be the same. This would still result in gaps for the inner walls as those you'd still want to print faster to keep print times down.

buzzhuzz, 2024-06-15:
Take a look at https://github.com/SoftFever/OrcaSlicer/pull/5609 which may solve all of your problems.

markleaf131313, 2024-06-15:
Oh cool, I didn't check the pull requests. I'll close this as that is pretty much what I was thinking of.

closed 2024-06-15 by markleaf131313