OrcaSlicer/OrcaSlicer #5714: Very slow performance

TheUnlimited64, 2024-06-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Im using a Surface Studio Laptop (1gen) and have 16gb of RAM

### Printer

Bambulab P1P

### How to reproduce

1. Click on the button where you can add custom filaments
2. Notice that Windows says Orcaslicer is unresponsive

### Actual results

Orcaslicer has very poor performance

### Expected results

Improved performance

### Project file & Debug log uploads

It's not project specific, because even a new project makes that, but ive included a log
[debug_Thu_Jun_06_18_53_00_21200.log.zip](https://github.com/user-attachments/files/15846223/debug_Thu_Jun_06_18_53_00_21200.log.zip)
The newest log could contain the freeze

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Could it be that the UI Thread and the Worker thread aren't disconnected, so the UI freezes while the Worker thread does its job?

L2on1, 2024-06-15:
Same issue here, my orcaslicer is very very slow and laggy

ASUS zephyrus G14 : Windows 11 / AMD Ryzen 7 4800HS / rtx 2060

I've already tried solutions from #3899

TheUnlimited64, 2024-06-16:
Same, its almost unusable waiting 30s before the custom filaments tab opens or 1-2min to open a project

pluciorx, 2024-06-25:
looks like duplicate of: #997

closed 2024-10-01 by github-actions[bot]

gthb96, 2025-11-15:
This is still a problem.
Also the application start is very slow.
I think it is because of Windows Defender.
When you first open Orcaslicer after Windows restart, Defender is very busy.

I suspect the reason is Orcaslicer binaries are not digitally signed.

You can look in taskmanager if Defender is doing heavy work while Orca is slow.