OrcaSlicer/OrcaSlicer #5717: OrcaSlicer bin on AUR has some issues

Skyslycer, 2024-06-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Manjaro 23.1.4

### Printer

Flashforge Adventurer 3 (can't test with other printers right now)

### How to reproduce

1. Open OrcaSlicer
2. Navigate anywhere and you will see the issue all over the place

### Actual results

Some text just doesn't appear. The whole application just feels "wonky".
For example, when clicking the settings icon on the printer settings, just an empty window.
Also the home and device tabs are empty aswell.

### Expected results

It should just work as expected and show everything as expected.

### Project file & Debug log uploads

The files just don't want to upload, I don't know why. The markdown text flashes for a second and then is gone again.
I was able to upload the images to an image uploader though: [https://postimg.cc/gallery/zTpqZLf](https://postimg.cc/gallery/zTpqZLf)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

It seems like I can upload everything here.
[image]
[image]
[image]
[image]
[image]
[image]

It still doesn't want to take the log zip.

kozross, 2024-06-15:
I have this problem as well.

[image]

This has been the case for _multiple_ releases now, and keeps changing on each release.

GraciousGpal, 2024-06-18:
Try setting the environment variable in console and running it from there:
`export WEBKIT_DISABLE_DMABUF_RENDERER=1`
then run Orca slicer.
Worked for me fixed blank screen.

kozross, 2024-06-19:
> Try setting the environment variable in console and running it from there: `export WEBKIT_DISABLE_DMABUF_RENDERER=1` then run Orca slicer. Worked for me fixed blank screen.

Just tried this, no different to my screencap above.

Skyslycer, 2024-07-14:
Hello, any updates?

kozross, 2024-10-13:
Any news on this issue?

Everspace, 2024-12-01:
I'm on Manjaro as well having problems starting Orca from the AUR. I get completely blank screens similar to the OP, including the setup dialogue (which can't be exited from the GUI and requires me to actively kill it from terminal).

```
Operating System: Manjaro Linux rolling
KDE Plasma Version: 6.2.4
KDE Frameworks Version: 6.8.0
Qt Version: 6.8.0
Kernel Version: 6.11.10-2-MANJARO (64-bit)
Graphics Platform: X11
Processors: 20 × Intel® Core™ i9-10900K CPU @ 3.70GHz
Memory: 62.6 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 3070 Ti/PCIe/SSE2
Manufacturer: ASUS
```

Running `orca-slicer` from terminal gets me a bunch of GTK-CRITICALs but more importantly:

```
src/nv_gbm.c:288: GBM-DRV error (nv_gbm_create_device_native): nv_common_gbm_create_device failed (ret=-1)

src/nv_gbm.c:288: GBM-DRV error (nv_gbm_create_device_native): nv_common_gbm_create_device failed (ret=-1)

src/nv_gbm.c:288: GBM-DRV error (nv_gbm_create_device_native): nv_common_gbm_create_device failed (ret=-1)

src/nv_gbm.c:288: GBM-DRV error (nv_gbm_create_device_native): nv_common_gbm_create_device failed (ret=-1)

KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
KMS: DRM_IOCTL_MODE_CREATE_DUMB failed: Permission denied
Failed to create GBM buffer of size 1200x732: Permission denied
```

However the `export WEBKIT_DISABLE_DMABUF_RENDERER=1` means I do not get the above problems and the application renders as normal.

Everspace, 2025-03-02:
It continues to be a problem for me.

closed 2025-06-09 by github-actions[bot]