OrcaSlicer/OrcaSlicer #5720: Сrashes on startup

sizzname, 2024-06-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta, nightly-builds, 6 hours ago, commit 1bb8fad

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

no matter

### How to reproduce

Run program.

### Actual results

Orca Slicer start. The main program window with menu is shown. It hangs and after some time the window disappears, the process is terminated.

### Expected results

It works

### Project file & Debug log uploads

**Debug log:**
```
[warning]	2024-06-15 22:23:39.028134[Thread 0x000044e8]:Failed to load to hints.cereal. File does not exists. C:\Users\s...\AppData\Roaming\OrcaSlicer\user\hints.cereal
[error]	2024-06-15 22:23:41.341630[Thread 0x000044e8]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
```

**Crash Log:** 
```
Exceptions:

Exception Code :c0000005 ACCESS_VIOLATION

Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x36d0f004 
Context :000000EA7FECD2C0 
ContextFlag : 0x10005f, EFlags: 0x10283 
Fault address:  0x80C2D703 0x1:0x1C8C703 C:\Program Files\OrcaSlicer\OrcaSlicer.dll

Show CallStack:
7ffe80c2d703:Unknown [Unknown file][-1]

Logical Address:
0x80C2D703 0x1:0x1C8C703 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
```

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sizzname, 2024-06-16:
New version (commit 1bb8fad) works fine.

HAndreasH, 2024-06-17:
had the same issue, works fine now

tlhintoq, 2024-06-19:
Ditto.  The beta was doing it to me as well - the RC seems to have fixed it

closed 2024-09-25 by github-actions[bot]