OrcaSlicer/OrcaSlicer #5724: detect overhang walls interferes with bridging

BETLOG, 2024-06-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Kubuntu 24.04

### Printer

frankenstein anycubic i3 mega, Ideaformer IR3 extruder, bambu nozzle

### How to reproduce

Create model with areas that are to be bridged.

### Actual results

Sometimes areas that are designed to be bridged get overhang wall treatment, and filled with a concentric layer.
This increases the chance of failure.
In many cases a symmetrical stl will have bridge assigned to one feature, and overhang wall assigned to it's mirrored twin, for no apparent reason.
In some cases this seemed to be accompanied by 'floating cantilever' warnings.

### Expected results

Probably bridging should always take precedence, and maybe query the user about what treatment to give otherwise.
Orca really needs a way to zoom on a specific area to display what some of its warnings are talking about.

### Project file & Debug log uploads

[anycubic-i3Mega_gantrySpoolMount_frame.zip](https://github.com/user-attachments/files/15856179/anycubic-i3Mega_gantrySpoolMount_frame.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

The log/project zip is not useful for demonstrating the seemingly random selection of overhang or bridge, as the first few times I noticed this all I did was take screenshots, and didnt save any 3mf.

Below: The 3mf infcuded here is merely to demonstrate what I believe is the cause of the issue. ie: if 'detect overhang walls' is on then it may interfere with things that are intended to be bridged.

[image]
[image]

Below: instances of mesh being sliced with a different treatment to otherwise identical overhangs, both intended to be bridged.

[image]
[image]
[image]
[image]

Below: 'bridge counterbore holes' with sacrificial layer failing 1/4 of the time, despite all 4 holes being essentially identical, and the left and right section of stl literally being mirrored. I was also seeing some interaction with 'detect overhang walls' here too.

[image]
[image]

tome9111991, 2024-06-16:
same issue like me
https://github.com/SoftFever/OrcaSlicer/issues/5706

I think it's a problem with the "extra perimeters on overhangs" setting

BETLOG, 2024-06-16:
> same issue like me #5706
 
agreed

BlackNet, 2024-06-16:
Been dealing with this same issue for the past 2 weeks.

tastyratz, 2024-07-06:
I was noticing the same behavior myself just now and found your report in a search.

Here is an example of very odd overhang detection displacing bridges on the temperature tower

[image]

I was originally trying to replicate and test https://github.com/SoftFever/OrcaSlicer/issues/4040 when I noticed the left edges were not even being detected as overhangs at all for me.

This also impacts line ordering. The "overhangs" are printed before the bridges (even though they are all actual bridges).

tastyratz, 2024-07-09:
I am running 2.1.1 and it was fine when I was on 1.9.x.

I thought I'd followup with another model link and real-life example here: 
https://www.printables.com/model/585445-voron-02r1-wide-feet-stock-skirts-update
os [image]

(For reference I also have this on the build plate: https://www.printables.com/model/185070-voron-misumi-2020-endcap )

it's really peculiar the way bridge and overhang detection is happening
os [image]
I'm seeing different behavior on the same plate with 2 parts. The misumi endcaps you can see are odd in the way they detect bridging and internal bridging because of the gaps over the voron logo and... well I can't answer WHY it would be leaving a gap an doing internal bridging after (must be some artifact with the model)
The v0 wide legs are showing overhangs over the built in supports which is more fitting.

For reference, I decided to try and slice the same part in the latest Superslicer beta just now 
ss [image]
it's properly detecting the area over the built in support as bridge infill

Curiously, superslicer does their overhangs very differently specific to the one that failed the most and it's NOT this way in the model so they must be handling regular bridges different.
ss [image]
ss [image]

Note too that the misumi endcap odd behavior in Orca doesn't seem to be present in SS by comparison
[image]

All that being said, sadly, the bridge bug ruined an otherwise perfect print trying to bridge on my overhang/logic settings.  I'll have to wait to print my V0 parts for this one to be fixed because I can't seem to get a clean abs bridge anymore on my trident XOL setup.
[image]

lightmaster, 2024-10-04:
On the X1C 0.4mm profile with a standard 3DBenchy loaded, if you change wall loops from 2 to 3 and turn on Extra Perimeters on Overhangs, it changes the cabin roof from a series of bridges to a concentric bridge. I've seen multiple people here say to turn off Extra Perimeters on Overhangs, but I think this is a bug with overhang detection, not a case of "just turn off X setting".

[image]

CCS86, 2024-10-07:
Fixes here would be very useful.

5p0rkz1lla, 2024-10-07:
I'm not sure if it will help find a solution, but I did discover something I don't see in this discussion. The issue appears to only apply when printing over another portion of the model. For the wall/bridge it's as if instead of just checking the previous layer it's checking all previous layers. When there's only the bed below things work correctly...

[image]
[image]

hotellonely, 2024-10-21:
this is still an active issue that needs to be solved. currently, the only "fix" is to change "Bridge Counterbore Hole" method to "Partially Bridged". Note that both None and Sacrificial Layer won't work. Only working one is Partially Bridged. Which doesn't make sense and can only be considered as a bug.

Example here:
[image]

lightmaster, 2024-10-28:
@hotellonely Wait, a bridge is supposed to look like that first image, where there are perimeters before and after the bridge but none on the non-anchored sides? I always thought it was supposed to look like that second image.

hotellonely, 2024-10-30:
> @hotellonely Wait, a bridge is supposed to look like that first image, where there are perimeters before and after the bridge but none on the non-anchored sides? I always thought it was supposed to look like that second image.

At least that's what Bambulab now thinks. They submitted a change as _a bug fix_, to change the behavior to look like the first image by default.

A bridging perimeter should be printed like a bridge rather than overhang because there's nothing that you can attach onto.

Slow overhang helps because it has something to attach onto from previously printed lines, but, because you don't have anything to cling on with the bridging perimeter when you're printing outter-inner or inner-outter-inner mode, so it should be printed as a bridge instead of overhang.

I think it all started from Prusa slicer which didn't come with OI or IOI print order previously, so they didn't really have a problem with that.

staal54a, 2024-12-09:
@hotellonely What's really strange is that the "fix" with setting the Bridge counterbore holes setting to "Partially bridged" seems to use the same routing that an internal bridge would but applies the external bridge speed.

As a note, I think Orca Slicer always will view a bridge as a 100% overhang per this comment: https://github.com/SoftFever/OrcaSlicer/issues/5788#issuecomment-2182741408

Can you try disabling Slow Down for Curled Perimeters and seeing if the bridging speeds are correct? When I do that, I still get the same line types with the outer perimeters detected as overhang walls and continuing around walls of the structure being bridged to, but the speed of those lines are correct.

Overall, I think there are several issues that may not necessarily be directly related but all stem from this overlap between bridges and overhangs that need to be addressed in some way. Related issues I have found so far:
https://github.com/SoftFever/OrcaSlicer/issues/6448 (feature request)
https://github.com/SoftFever/OrcaSlicer/issues/2231
https://github.com/SoftFever/OrcaSlicer/issues/6941 (this one apparently has the counterbore partially bridged option converting actual overhangs to bridges)

There may be more but it really does seem like bridges and overhangs have some underlying issues.

hotellonely, 2024-12-17:
> @hotellonely What's really strange is that the "fix" with setting the Bridge counterbore holes setting to "Partially bridged" seems to use the same routing that an internal bridge would but applies the external bridge speed.
> 
> As a note, I think Orca Slicer always will view a bridge as a 100% overhang per this comment: [#5788 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/5788#issuecomment-2182741408)
> 
> Can you try disabling Slow Down for Curled Perimeters and seeing if the bridging speeds are correct? When I do that, I still get the same line types with the outer perimeters detected as overhang walls and continuing around walls of the structure being bridged to, but the speed of those lines are correct.
> 
> Overall, I think there are several issues that may not necessarily be directly related but all stem from this overlap between bridges and overhangs that need to be addressed in some way. Related issues I have found so far: #6448 (feature request) #2231 #6941 (this one apparently has the counterbore partially bridged option converting actual overhangs to bridges)
> 
> There may be more but it really does seem like bridges and overhangs have some underlying issues.

I'm not sure if I have stated it here or somewhere else before, but no turning off "slow down for overhang" is not a solution to this problem. I've made various tests and extra perimeter, slowdown for overhang,... none of them can provide consistent fix for this issue. 

The only rather consistent fix is the counterbore hole, by changing it to partially bridged.

kayboy-bebop, 2024-12-22:
> > @hotellonely What's really strange is that the "fix" with setting the Bridge counterbore holes setting to "Partially bridged" seems to use the same routing that an internal bridge would but applies the external bridge speed.
> > As a note, I think Orca Slicer always will view a bridge as a 100% overhang per this comment: [#5788 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/5788#issuecomment-2182741408)
> > Can you try disabling Slow Down for Curled Perimeters and seeing if the bridging speeds are correct? When I do that, I still get the same line types with the outer perimeters detected as overhang walls and continuing around walls of the structure being bridged to, but the speed of those lines are correct.
> > Overall, I think there are several issues that may not necessarily be directly related but all stem from this overlap between bridges and overhangs that need to be addressed in some way. Related issues I have found so far: #6448 (feature request) #2231 #6941 (this one apparently has the counterbore partially bridged option converting actual overhangs to bridges)
> > There may be more but it really does seem like bridges and overhangs have some underlying issues.
> 
> I'm not sure if I have stated it here or somewhere else before, but no turning off "slow down for overhang" is not a solution to this problem. I've made various tests and extra perimeter, slowdown for overhang,... none of them can provide consistent fix for this issue.
> 
> The only rather consistent fix is the counterbore hole, by changing it to partially bridged. it doesn't work perfectly, but well enough or at least better than the wonky output...

This definitely fixes some things, but I think that's largely down to the counterbore hole option replacing so much of the normal sliced gcode with the feature specific gcode (which sometimes introduces it's own problems). In areas of the sliced model where counterbore doesn't activate I still have bridges being treated as overhangs.

But really, overhangs being jank has been an issue with all Prusa Slicer forks for a while. I've hopped between Prusa, Orca and Super trying to consistent overhang/bridge behaviour, but at thsiI can only hope that one of the next big releases is focused on bug fixes, cause a lot of these new fancy features (as great and exciting as they are) are breaking things.

digital0785, 2025-08-06:
> [@hotellonely](https://github.com/hotellonely) What's really strange is that the "fix" with setting the Bridge counterbore holes setting to "Partially bridged" seems to use the same routing that an [...]

the problemm with the partially bridged "fix" is that it also breaks things .. 

this is no  partially bridge which is printing concentric however the walls stay proper

[image]

[image]

this is turning on partially bridged obviously fixes the concentric issue however creates gaps in layers.. it clearly could print a perimeter around the bridge however it doesn't which creates an empty layer where the bridging happens. the bridge should only be over lapping 1 wall as an anchor which it does most of the time but for some reason it breaches the outer wall in 1 spot and doesn't add the wall on the rest

[image]

[image]

this happens both with arachne and classic as well and I've tried going down to 5% minimum wall thickness and it's still an issue

andrewleek, 2025-08-29:
Howdy everyone,
I have made a PR regarding overhangs and bridges and it should solve some of the issues in this thread.

It would be great to get some testing and feedback to make sure it works correctly and there arent any unforeseen side-effects.

PR:
https://github.com/SoftFever/OrcaSlicer/pull/10565

You can download the builds under "artifacts":
https://github.com/SoftFever/OrcaSlicer/actions/runs/17304272227?pr=10565

Thank you