OrcaSlicer/OrcaSlicer #5726: Slicing Crash (Unsure of cause)

hYdos, 2024-06-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

CPU: 13th gen 19-13900ks
GPU: RTX 2070 Super
RAM: 32g (ram usage never went over 60%)

### Printer

Modified Ender 3

### How to reproduce

1. Open 3mf project
2. Attempt to slice
3. Crash

### Actual results

Orca slicer freezes for a second or two then crashes.

### Expected results

File should be sliced.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15858839/log.zip)
[crash.zip](https://github.com/user-attachments/files/15858840/crash.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

hYdos, 2024-06-16:
TreeSlim seems to be the cause.

andreaaraya01, 2024-06-20:
I have same problem,I though the Honeycomb pattern was the problem but is happening with treeslim and rectilinear too, this happen to me only with complex figures

hYdos, 2024-06-20:
Same, I found it only happens on complex models

fernandoavf7, 2024-06-20:
These are the support parameters that cause freezing in 90% cases with complex models like figures, I tried the same parameters in a Ryzen 5800X + GTX1080 and with apple M1 CPU, so I think is noi a PC problem, I tried to slicing with similar support options in Cura and get the same freezing, but in Creality slicer works.

[image]

andreaaraya01, 2024-06-28:
I update to version 2.1.1 but problem persist yet

hYdos, 2024-06-28:
Can confirm same for me

fernandoavf7, 2024-06-28:
I tried in last version with rectilinear and works ok, but with honeycomb pattern stuck at 50% [image]

hYdos, 2024-09-27:
Just because the maintainers are lazy does not make this stale

closed 2025-01-03 by github-actions[bot]