OrcaSlicer/OrcaSlicer #5728: No wiki and no settings explanations

otede, 2024-06-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

 Intel i3-12100F

RTX 2070 8GB MSI

64 GB RAM

Motherboard MSI PRO B660M-A WIFI DDR4 (MS-7D43) 

### Printer

Bambu Lab A1

### How to reproduce

1. Go to https://github.com/SoftFever/OrcaSlicer/wiki/
2. Click on any of the 3 links there

### Actual results

Links not opening any new pages. The site reloads.

### Expected results

A wiki of some sort maybe...

Not to mention Cura has the stellar Settings Guide Extension.

### Project file & Debug log uploads

[heart infinity projektorca 02b.zip](https://github.com/user-attachments/files/15859416/heart.infinity.projektorca.02b.zip)
[debug_Sun_Jun_16_20_09_26_14060.log.0.zip](https://github.com/user-attachments/files/15859420/debug_Sun_Jun_16_20_09_26_14060.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

To top that off, there are some really odd defaults, e.g. max flow rate in PLA Generic is 12, support defaults result in parts straight to trash. But that's a side not. I'd like to get to the wiki (if it exists) first.

Ohmarinus, 2024-06-16:
How is this a bug in the software? Please don't abuse the bug tag if you're filing a complaint (or feature request).

Instead use the "Feature Request / Enhancement" option: https://github.com/SoftFever/OrcaSlicer/issues/new?assignees=&labels=enhancement&projects=&template=feature_request.yml

For now you can hover over the settings in Orca to get a short explanation of what the settings do. Not everything has an explanation yet as some functions are quite new. I agree that it's important to get a better wiki and working explanations for all the functions but the [Feature Request](https://github.com/SoftFever/OrcaSlicer/issues/new?assignees=&labels=enhancement&projects=&template=feature_request.yml) path is the best path to follow in that case :)

otede, 2024-06-16:
Sorry, I didn't notice any other tags along the way. I WILL use the "feature" option in the future. Can you edit this entry for now?

As for the wiki, I don't think it's a matter of a "better wiki". If the presented links are all there is then Cura wiki is rather non-existent, apart from maybe the calibration article you have to hack your way into. Also, there are clickable links all over Cura's interface. All with dead links leading to https://github.com/SoftFever/OrcaSlicer/wiki/ in the end.

Just to confirm, could you please "convert" this entry to a feature request?

otede, 2024-08-08:
A proper entry in Enhancements, this one can be closed
https://github.com/SoftFever/OrcaSlicer/issues/5822

closed 2024-11-16 by github-actions[bot]