OrcaSlicer/OrcaSlicer #5731: Add maximum volumetric rate to printer settings

jcleclaire, 2024-06-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Maximum extrusion rate is typically limited by the hotend nozzle size. However, the place where this is set (printer settings) does not have a field for the maximum volumetric rate.

The maximum rate should be specified in the printer settings as it is a maximum that is largely limited by these factors.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add an input field under _Printer > Extruder_ for _Maximum Volumetric Rate_ spied in mm^3/s. The volumetric flow rate should use the lesser of this value and the value specified in the specific filament profiles.

It may also be useful to allow the maximum volumetric flow rate specified under filament settings to be expressed as an % which would be calculated of the printer's maximum volumetric rate.

### Describe alternatives you've considered

Creating redundant filament profiles for different sized nozzles/extruders (ie Rapido HF vs UHF).

TheUnlimited64, 2024-06-17:
I wouldn't do that because its often very filament/temperature specific. Meta PLA from Sunlu in my tests can archive 40mm3/s, but TPU cant be printed like that

jcleclaire, 2024-06-17:
@TheUnlimited64 I think it is still valuable to have it, as w change nozzles from a 0.4 to a 0.6CHT your flow rate will increase considerably. Having multiple filament profiles for this is inefficient.

You can still have filament profiles with volumetric limits for edge-cases like TPU (which may also be solved with a specific print settings profile), but you can also leave it blank in the filament flow rate and have the slicer use the default printer volumetric rate.

jcleclaire, 2024-06-19:
@mjonuschat reading through the discussion I can understand their point of view.

One thing that didn't get mentioned explicitly is that this is a 2D problem, 1 dimension in the filament and it's properties, the other is the printer's capabilities; however only the filament properties is being captured. Maybe an intermittent solution such as under printer properties a series of values for different filament types (ie one maximum volumetric rate for TPU, one for PLA, one for ABS, etc). This would allow the 2D combination of filament type + printer properties in a single space.

Thoughts?

florek41, 2024-06-22:
#3972 @SoftFever 
The need for varying Max Volumetric Speed is similar to retraction settings. For example, PLA with a Bowden setup might require even 5mm retraction, whereas TPU would need no retraction, or print faliure occurs, hence we can override this in the filament profile. The same principle applies to Max Volumetric Speed. Different printers and hotends require different settings for optimal performance and safety, making an override feature essential.

closed 2024-09-28 by github-actions[bot]