OrcaSlicer/OrcaSlicer #5732: Delayed printing function

rovesoul, 2024-06-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Not involving existing issues

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I have a need that I believe many others may also have. Assuming I have a large family including elderly people, children, and pregnant women who are sensitive to noise, I hope to add a delay function when sending print jobs to the printer. This way, the printer can start working automatically

TheUnlimited64, 2024-06-17:
Would mean your PC has to be on I guess, otherwise it couldn't send it. I believe it would be smarter to integrate such feature first in the printer software, so the printer itself can schedule the prints itself

rovesoul, 2024-06-18:
> Would mean your PC has to be on I guess, otherwise it couldn't send it. I believe it would be smarter to integrate such feature first in the printer software, so the printer itself can schedule the prints itself

No this satiation ，I just want delay print object,just want when no one at home,the machine work

foreachthing, 2024-06-18:
How about that: https://marlinfw.org/docs/gcode/G004.html
Wait 60 seconds `G4 S60`.
I guess you'll need to try it first ;-)

rovesoul, 2024-09-17:
您的来信已经收到,谢谢。待我尽快上线处理。
                                           
                                       ——董汇标