OrcaSlicer/OrcaSlicer #5733: can't upload to sd card  only print via cloud  bambu x1c

broncosis, 2024-06-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta

### Operating System (OS)

Linux

### OS Version

Linux mint 21.3

### Additional system information

Cpu ryzen 7 3700x 32 gis of ram 

### Printer

Bambu x1c 

### How to reproduce

slice model click arrow choose send rather than print 
then click send 

### Actual results

Error code : -5010
Error desc: Failed to upload print file to FTP please check network status 
Extra info : [ftp code]: 3 [reason]: URL using bad/illegal format or missing url 

### Expected results

it should send the file to the sd card 

works as expected with bambu studio 

### Project file & Debug log uploads

[image]
[voronkeystone.zip](https://github.com/user-attachments/files/15862167/voronkeystone.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]
[voronkeystone.zip](https://github.com/user-attachments/files/15862167/voronkeystone.zip)

doudar, 2024-08-09:
I'm getting this exact same bug on windows. Bambu Studio sends the files okay.

doudar, 2024-08-19:
@broncosis , were you able to find a solution to this? I've completely re-installed windows and OrcaSlicer since this happened, and I still can't send files to the SD card.

closed 2024-11-26 by github-actions[bot]