OrcaSlicer/OrcaSlicer #5738: Allow marking specific layers to be treated as full "top layers"

stew675, 2024-06-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When printing objects that have large flat areas with additional surface features on top, the slicer will try to extend the infill vertically into the top features.  This results in the layer that the features sit on top of having an inconsistent appearance as the slicer has to leave gaps in the surface for the lower infill to extend vertically into the upper features.

Embossed text would be one such example where this occurs, where the layer that the text sits atop of is full of moves to exclude the areas that the embossing sits on because it wants to extend the infill vertically into the embossing.  The action of filling in the broken up surface often results in a messy appearance.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Ideally, much like how we can right-click on a layer height to add in custom g-code, there would be an option to say "Treat as a top layer".

Basically the slicer would treat the object up to that layer as if it were a complete object in its own right, and additional features on top of that as before.

The best way to describe this would work would be like the following gcode splicing example:

- Slice an object normally.
- Now cut the object at the layer that you want to have a clean appearance
- Reslice the object up to the cut as if it were a complete object with its top layer at that height.
- Remove the PRINT_END portion of the lower-cut's gcode
- Now insert the original g-code from above that cut point onto the end of the layer-cut gcode

Now we have generated g-code that has a fully clean middle-layer that has features added atop of it, without the broken surface moves to avoid the infill that used to extend between the upper and lower portions.

This will result in much cleaner prints without having to do the above g-code splicing work by hand.

jeremytodd1, 2024-06-17:
I think I know what you're asking for and I believe the feature you _actually_ want is SuperSlicer's "**Anchor Solid Infill by X mm**" setting

This specific feature is a better solution to this issue, and is the one feature that I have desperately wanted for Orca to get for about a year now. This single setting has huge impacts on top layers for some models, especially for things like signs with text on the top, or even maps that people have been printing lately.

Here is a bit of an example to illustrate what "Anchor Solid Infill by X mm" does.
This is a of "The Office."
[image]

After tuning that Anchor Solid Infill by X mm setting, this is how SuperSlicer prints the top layer before the top text:
[image]

Orca does not have this feature at all, so it's impossible to tune for this. This is how Orca prints the same layer.
[image]

SuperSlicer is able to print the entire layer in one go, which makes it look much more consistent.
Orca has to stop the line and continue it again, which can cause inconsistencies in the look of the layer.

@SoftFever , @igiannakas ,
TLDR: Having this Anchor Solid Infill by X mm feature would be a great way to keep the top layers looking consistent all across the board.

Ataraxia-Mechanica, 2024-06-18:
You can add a height range modifier to that layer and the layer below it, set it to 0 top layers and 100% infill, disable only one wall on top, then slow solid infill down so its speed matches top layer. Very hacky but it works.
[image]
[image]

jeremytodd1, 2024-06-18:
> You can add a height range modifier to that layer and the layer below it, set it to 0 top layers and 100% infill, disable only one wall on top, then slow solid infill down so its speed matches top layer. Very hacky but it works. 

Thank you for this. This is the only workaround that I've seen that works. (Just to point out, you can replicate the "only one wall on top layer" setting by just simply changing wall count to 1 in the modifier as well, so you don't need to lose that setting for the top layer)

Saying that, I think this only shows how badly Orca needs the **Anchor Solid Infill by X mm** setting. That is a lot of steps instead of just tuning one simple value.

stew675, 2024-06-19:
> I think I know what you're asking for and I believe the feature you _actually_ want is SuperSlicer's "**Anchor Solid Infill by X mm**" setting

Thank you for that.  The videos you provided are pretty much exactly that.  The Orca Slicer video perfectly explains what Orca Slicer does today, and it's really annoying.

I'm not particularly fussed by what name it would go under, but I think it's important that the layer gets treated as a top layer, rather than as Solid Infill.  Perhaps SuperSlicer is treating it as a top layer though?

> Thank you for this. This is the only workaround that I've seen that works. (Just to point out, you can replicate the "only one wall on top layer" setting by just simply changing wall count to 1 in the modifier as well, so you don't need to lose that setting for the top layer)

Agree.  Thank you for the work around @Ataraxia-Mechanica but that is way more complicated.

> Saying that, I think this only shows how badly Orca needs the Anchor Solid Infill by X mm setting. That is a lot of steps instead of just tuning one simple value.

I'd even be happy with a radio button that simply says "Treat layers that have partial top-layer infill as a complete/full top layer", so that way it catches all layers that have top-surface features without the user having to manually fiddle with it.  The subsequent (higher) layers should start off as if there were 0 bottom layers (ie. immediately start printing infill in the next layer just like what happens currently).

jeremytodd1, 2024-07-05:
@Ataraxia-Mechanica 

Another issue with that solution is that with the modifier, it's no longer considered a top surface.
[image]

Due to this, Lift Z Enforcement's "Top and Bottom" surfaces setting will not work on this. This will cause scratches on the top layer. You can see here there is no lift in the travels:
[image]

vovodroid, 2024-07-26:
@jeremytodd1 , what is exact option name for **Anchor Solid Infill by X mm**? I don't see it in SS 2.5.59

jeremytodd1, 2024-07-26:
@vovodroid 

It looks like the actual value's name is "external_infill_margin"
[image]

You can find it in the Infill section in the Print Settings tab.

Eldenroot, 2024-07-26:
I believe @vovodroid will send a new PR for this soon :)

jeremytodd1, 2024-07-26:
That would be absolutely amazing. Not being able to tune for this has been haunting me ever since I switched to Orca. I really appreciate it.

vovodroid, 2024-07-27:
It doesn't work now in SS 2.7 branch, as function where it was implemented was changed since Prusa 2.5. Let's hope Merill will fix it eventually, thus it could be ported to Orca.

Anollya, 2024-09-25:
> That would be absolutely amazing. Not being able to tune for this has been haunting me ever since I switched to Orca. I really appreciate it.

I fully agree!
any news here?

old:
[image]

workaround - perfect:
[image]

vovodroid, 2024-11-06:
Try to set Ensure Vertical Shell Thickness to All, see #7386

jeremytodd1, 2024-11-24:
> Try to set Ensure Vertical Shell Thickness to All, see #7386

It doesn't seem to do a whole lot for me.
My latest test file is small map I made just for testing for this issue: 
https://www.printables.com/model/1084726-test-map-for-top-layer-issue

The goal of the test model is to have layer 6 (at 1.2mm height) print in just one clean layer that doesn't have any additional details on it. Layer 7 should be the layer where buildings and roads start printing.
Here is layer 6 for reference:
[image]

I see Prusa released an alpha version of PrusaSlicer 2.9. Do you know if this version would help improve things?

Nilssonfpv, 2025-02-03:
Still no fix for this short of the workaround that means you have no top players at all?
We could really do with a way to mark what the "top" layer is that text/logos will be printed ontop

Id be fine if it means id just have to have the model as assembled parts, where you mark them as  "top" or "base" parts

CodyPrince, 2025-03-20:
Any new on this? Would really like to see it in orca.

vovodroid, 2025-03-25:
@jeremytodd1 it seems to work in latest Superslicer 2.7.61.

[image]

[image]

[image]

[image]

Could you check this? If it was fixed there, I'll take a look on SS code and probably try port in Orca.

jeremytodd1, 2025-03-25:
@vovodroid 

I actually helped with testing for this specific issue with the SuperSlicer devs to help get this fixed in their latest release.
Yes, I believe it has been fixed in this latest version of SuperSlicer. I've tested it with numerous models at this point (technically on an older pre-release version but it it should act the same on this new release version) and they all printed with a nice clean main layer.
Layer 11:
[image]
Layer 10:
[image]

I've found I need two specific settings to be set correctly for it to work.

Ensure Vertical Shell Thickness needs to be set to "Enabled (2.5)". It will not work with the other settings.
[image]

Anchor solid infill by X mm is still necessary.
[image]

I typically leave the anchor solid setting to a value of "20" (not a percentage, just a hard value of 20) and it seems to work for most models. This value can differ though. For this office sign for example, I can set that value as low as 6 and the main layer will still print clean. If I set it lower, like to even at 5, it'll start showing artifacts in the main layer.

I would be thrilled if this was now able to be ported to OrcaSlicer. This issue still **very** real to me in Orcaslicer and would love for it to be fixed.

jeremytodd1, 2025-04-10:
@vovodroid Just curious, did you happen to look into this yet?

vovodroid, 2025-04-10:
Still no, but certainly will.

vovodroid, 2025-04-20:
Well, it's turn out to be more complicated than I guessed, as SS code base is different from Orca. I built ported code, but it broke functionality.

Here is the branch, may be somebody could help https://github.com/vovodroid/OrcaSlicer/tree/infill-margin-wip

@igiannakas ?

metinkale38, 2026-03-05:
I found a very simple workaround for the Emboss Tool in Orca:
 
You can set the property "From Surface" to 0.02 (lower values will be rounded to 0) and it will print the top layer without interruptions.