OrcaSlicer/OrcaSlicer #5740: Add ability to remove default color from filament settings (Windows)

markaudacity, 2024-06-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There doesn't seem to be a way to remove a default color once one is selected in v1.9.1 on Windows. I tried right-clicking on the color button and clicking it, closing the color picker, and hitting Delete while the color button is highlighted, but neither seems to work. The Windows color picker does not let you select no color.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Some method of removing the default color once assigned. I realize this is somewhat limited by the archaic Windows color picker dialog; perhaps highlight the color button and hit Delete? 

### Additional context

[image]

SoftFever, 2024-06-18:
For now, you will have to go to %appdata%/OrcaSlicer/user folder and find your filament profile(it's json file)
Then delete the `default_filament_colour` entry

    "default_filament_colour": [
        "#800040"
    ],

closed 2024-09-25 by github-actions[bot]

adobni, 2025-01-19:
Can this issue be reopened?

BlasterGales, 2025-01-29:
please someone with permission reopen this

FHEMUser, 2026-11-16:
With this line I have a problem with one color and Centauri Carbon 1 with canvas. If this line ist not 
standard_filament_color = ""
my printer ask always for 2 colors if I print with one.

I try many things and install the Elegoo Slicer. Here this line is like I wrote and then only one spool will be shown at starting.

I am the only person with this ?
I think if I select both spools the same color it will work, don't try it but it's not fine