OrcaSlicer/OrcaSlicer #5746: Improve overhang performance with adaptive outer wall line width

Ataraxia-Mechanica, 2024-06-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When printing objects with steep overhangs, using a thicker outer wall line width could increase overhang performance. This is due to more materials being deposited on top of previous layer.

[image]

But thicker outer walls make sharp details look worse. I wonder whether if it would be possible to detect overhangs and make them thicker, but make non-overhang areas use the original line width.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

[image]

Detect overhangs and use thicker outer wall width on them, while not changing non-overhang wall line width.

vgdh, 2024-06-18:
That would be a great improvement!
I wonder what @SoftFever @Noisyfox @igiannakas  think about it.

igiannakas, 2024-06-18:
Without trying it we can’t say for sure what it will do, but from first looks it will improve step over and overhang performance for sure - I personally run 0.48 external wall line width in my profiles and 0.42 internal.

However varying line widths on a single surface may cause appearance inconsistencies - my theory is that you want the extrusion to be as consistent as possible between layers. 

So I’m presuming that the other side of the overhangs will appear slightly worse than if you just kept a larger line width throughout all the external walls.

But won’t be sure unless we test it!!

Noisyfox, 2024-06-18:
> However varying line widths on a single surface may cause appearance inconsistencies

And also make it harder to implement too. Pretty much like traditional wall generator with a thicker wall width vs Arachne.

Ataraxia-Mechanica, 2024-06-18:
>However varying line widths on a single surface may cause appearance inconsistencies
I'm thinking about only increasing line width for the overhang part, then revert to normal width immediately after. Could probably still cause inconsistencies and artifacts on the corner of the width change though.

Ataraxia-Mechanica, 2024-06-18:
I also think this feature should only be enabled when it comes to extreme overhangs that can't be handled normally, e.g. 60 degree overhangs at 0.2mm layer height, using 0.8mm line width to make it printable.

SoftFever, 2024-06-18:
I doubt this would work.
The unsupported part remains unchanged, and your nozzle position remains unchanged.
Only changing layer height would reduce the unsupported part

Ataraxia-Mechanica, 2024-06-18:
[image]

One test, coming up!
Hope the aux fans I installed on both side have the same power...

Ataraxia-Mechanica, 2024-06-18:
Currently suffering from success. Both cases worked even with aux fan off. Just gimme a bit more time...

igiannakas, 2024-06-18:
Up to a point it does - there is a big difference going from 0.4 to 0.48 in my experience. 

After that it starts tapering off and above 0.5-0.6 or so it starts creating other artefacts as the overlap isn’t as much and the plastic just gets pushed out on the overhanging side if that makes sense. 

Personally I run all my print profiles with 0.45-0.48 external perimeters for this reason - to get good step over for inner outer inner wall ordering. But I haven’t felt the need to dynamically change this as the fixed line width works well enough without any consequences. It also increases part strength quite nicely.

Stefan had a video showing line width impacts on print quality here: https://www.cnckitchen.com/blog/the-effect-of-extrusion-width-on-strength-and-quality-of-3d-prints?format=amp

Ataraxia-Mechanica, 2024-06-18:
At extreme angles, thicker line width actually has performs worse on my printer, as the thicker lines are easier to sag down and getting separated from inner walls, while 0.4mm lines literally drips down, gets blown by the fans, shrinks and grows back up, then sticks on the lower side firmly. However, I think I already mentioned a situation that makes thicker walls strictly better 2 posts before: concave overhangs.
[image]

Ataraxia-Mechanica, 2024-06-18:
[image]
Guess which one is 0.8mm outer wall and which is 0.4

closed 2024-09-26 by github-actions[bot]