OrcaSlicer/OrcaSlicer #5747: Bridges don't slice correctly - Bridges don't anchor to perimeters correctly.

wbagnall, 2024-06-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Voron 2.4

### How to reproduce

I just spend an hour writing this up and getting all the screenshots, and then the page decided to refresh and reset all the prompts. Oh well. Here I go again.

To reproduce, slice the project files I uploaded. I have played around with various settings to try and fix the issue, but with no positive results. I cannot guarantee that every single profile will produce these results, but it seems to happen no matter what settings I change. 

### Actual results

As said in the title, bridges don't anchor to perimeters correctly, leaving empty gaps or gaps of (imo useless) sparse infill instead of fully anchoring to all the perimeters. Occasionally bridges decide not to anchor to anything at all.

In this example, you can see the bridges don't anchor completely over the perimeters. See The Expected results section for what should happen.
[image]

In this example, bridges print over nothing (as well as not completely anchoring over perimeters). See the expected results section
[image]
"ksr_fdmtest_v4.3mf"

In these examples, bridges don't anchor correctly, causing the bridge to fail.
[image]

[image]

[image]
"incorrect bridging.3mf" "incorrect bridging 2.3mf"

In this example of the OrcaCube, the bridges don't completely anchor, leaving a gap for the next layer.
[image]

In this example of the 3D Benchy, a bridge is generated where an overhang should be
[image]

Here you can see the incomplete anchoring with the sparse infill. I believe this is intentional, but it reduces bridge quality because the lines are only anchored to 1 perimeter instead of completely.
[image]

In this temp test, you can see the bridge isn't anchored completely, leaving a gap for the next layer.
[image]

### Expected results

Bridges should anchor entirely to perimeters without leaving any gaps for the next layer. The area of the gaps and weird sparse infill should be bridge anchors instead to improve bridge quality and reduce chance of failure. I suggest adding a setting that allows the user to select how many perimeters a bridge anchors into. Bridges should also anchor to _something_ instead of being printed mid air.

Here is an example of OrcaSlicer correctly slicing bridges without leaving any gaps or printing over nothing. "correct bridging.3mf" layer 4.
[image]

In regards to this case, the bridges need to be generated in such a way that they correctly attach to a perimeter, roughly like this.
[image]

### Project file & Debug log uploads

[Project Files.zip](https://github.com/user-attachments/files/15881429/Project.Files.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

wbagnall, 2024-06-18:
Here is how prusaslicer handles the ksr_fdmtest file.
[image]

wbagnall, 2024-06-18:
Here is how PrusaSlicer handles the "incorrect bridging.3mf" file. You can see that it properly anchors the bridges to the perimeters.
[image]

foreachthing, 2024-06-18:
One thing I've found with "incorrect bridging.gcode" -> **Ensure vertical shell thickness: All**

[image]

Compare to your screenshot:
[image]

Ataraxia-Mechanica, 2024-06-18:
You have bridge counterbore holes set to partial which doesn't play well with holes that aren't counterbored:
[image]

This doesn't fix the long bridging on the outside though, you can offset a part of it to be a layer higher to break up the bridging area.

SoftFever, 2024-06-18:
You will need to make some manual adjustment for some models.
For example, it would be better to set the bridging direction to 135 degree for the following model
Also, I suggest start with default settings, a lot of advanced features in Orca have very specific use cases.
It's not recommended to have them on if you don't need it. E.g. we don't bridge counterbore feature on in this case it you don't have counterbore sink holes and this will interfere normal bridges.
[image]

[image]
[image]

closed 2024-06-29 by wbagnall