OrcaSlicer/OrcaSlicer #5754: Camera doesn't work on ALL 2.1 versions, Beta and RC but work fine on 2.0 and Bambu Studio.

TexasTexasTexas, 2024-06-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0 Release Candidate

### Operating System (OS)

Windows

### OS Version

Windows 10 22H2

### Additional system information

AMD Ryzen 9 3900x
RTX 2070 Super
64GB DDR4 3200mhz

### Printer

Bambu Lab X1 Carbon

### How to reproduce

After installing Orcaslicer 2.1.0 Beta and Release candidate, I get this issue after downloading the Bambu Network Plugin. I am not sure how to reproduce it, i've reinstalled Orcaslicer 3 times and I cannot get it working on 2.1 but it works fine on 2.0.

### Actual results

[image]
"Please check the network and try again. You can restart or update the printer if the issue persists. [2: - 10]"

### Expected results

The slicer should refuse to load the camera, and will throw the above error each time you click the play button

### Project file & Debug log uploads

Happens on all projects, no crashes or anything.

GitHub wont let me upload the 0 file debug.

Here is the raw text:
[warning]	2024-06-18 13:59:10.537501[Thread 0x0000054c]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]	2024-06-18 13:59:10.538001[Thread 0x0000054c]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_profile_3mf
[error]	2024-06-18 13:59:17.254522[Thread 0x0000054c]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-06-18 13:59:17.268522[Thread 0x00003200]:Unable to open the fileC:\Program Files\OrcaSlicer\resources/calib/tolerance_test/OrcaToleranceTest.stl
[error]	2024-06-18 13:59:17.372040[Thread 0x00006768]:Unable to open the fileC:\Program Files\OrcaSlicer\resources/calib/volumetric_speed/SpeedTestStructure.step
[error]	2024-06-18 13:59:20.102016[Thread 0x0000054c]:Slic3r::GUI::PartPlate::calc_exclude_triangles:Unable to create exclude triangles
[error]	2024-06-18 13:59:20.324555[Thread 0x0000054c]:Slic3r::NetworkAgent::connect_server error: network_agent=000001EEA46902F0, ret=-2
[warning]	2024-06-18 13:59:20.628852[Thread 0x00000bec]:[Orca Updater]: online version=1.9.2.6, current_version=1.9.2.6, no need to download
[error]	2024-06-18 14:01:20.347774[Thread 0x0000054c]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 
[error]	2024-06-18 14:01:39.368417[Thread 0x0000054c]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 
[error]	2024-06-18 14:02:19.061319[Thread 0x0000054c]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 
[error]	2024-06-18 14:03:00.276743[Thread 0x0000054c]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 
[error]	2024-06-18 14:03:12.773983[Thread 0x0000054c]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 
[error]	2024-06-18 14:03:18.785304[Thread 0x0000054c]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 
[error]	2024-06-18 14:03:22.119422[Thread 0x0000054c]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 
[error]	2024-06-18 14:03:29.124731[Thread 0x0000054c]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 
[error]	2024-06-18 14:03:43.060638[Thread 0x0000054c]:Slic3r::GUI::Tab::tree_sel_change_delayedcan not find page with current selection 
[warning]	2024-06-18 14:03:44.305241[Thread 0x00004720]:SupportSpotsGenerator does not currently handle different materials properly, only first will be used

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Nothing else to add, aside from that it works with zero issues on Bambu Studio also.

W1SPS, 2024-06-22:
I have the same problem. I had it in the beta also, I thought the RC would fix it, but it didn't.  Guess I'll uninstall and go back until there's a fix.

texeryone, 2024-06-29:
Other people still having issues with it on the newest 2.11 download, any update?

mydevpeeps, 2024-07-04:
### Problem:
I just updated and after installing the plugin I also can't see my cameras. I can print ok to the printer, sync filaments and all that.. just no camera.. I get the same error as everyone else. 

### My Workaround Steps (Successful)
1. Updated Orca to 2.1.1.Got messages about plugin and BambuSource. Followed prompts. Video not working since ugrade.
2. Installed Bambu Studio 1.8. Followed all prompts to fix/install things. Orca still broke. Camera works in Bambu Studio.
3. Bambu Studio prompts me to get the latest version, 1.9, so I downloaded it and installed it. Followed all the prompts to install plugin but no error about BambuSource. Camera still working in Bambu Studio.
4. Opened Orca, ignored BambuSource error. Camera works now.

### Troubleshooting Details
I installed Bambu Studio 1.8 which then prompted me to update to 1.9.. not yet. I want to test. I installed the network plugin as prompted. It threw an error at me about "Using a BambuSource from a different install, video play ma not work correctly! Press Yes to fix it." Now, while this error is up, I can see the video going in the background.. I click Yes to fix it.. It states registration is successful. Just a note, all these steps had also happened in Orca before I went here to try this out. 

I close Bambu and open up Orca. I go to the video tab and I get the same error message, "Using a BambuSource from a different install, video play may not work correctly! Press Yes to fix it." Hrm. Whack-a-mole? I click no for now. Exit Orcla.

I decide to install the latest Bambu Studio it had me download (1.09) and see if maybe it has an updated version of something related to this. After upgrading and then opening bambu studio, it installs the network plugin again... I go to the Device tab and open up the camera feed. I don't get errors this time and it seems to work. Will this fix Orca?

I open up Orca again and hit the device, and play on the camera. The camera comes up but I get this error yet again, Using a BambuSource from a different install, video play may not work correctly! Press Yes to fix it." I click No because, well, it seems to be working now? I wonder if I need to say yes to that fix message or leave it alone and just always say no?

pgifford, 2024-07-04:
Works for me and has worked for me from 2.1.0-dev (see [my comment](https://github.com/SoftFever/OrcaSlicer/issues/5170#issuecomment-2106519565) on #5170) to the latest nightly of 2.1.1 which I installed earlier today. I’m running Windows 10 and have the latest Bambu Studio release installed.

drizztguen77, 2024-08-01:
> Other people still having issues with it on the newest 2.11 download, any update?

I'm still getting the error on 2.1.1. If I answer No it seems to work fine and I can see the camera. But it's concerning as if I accidentally hit Yes it will break Bambu Studio.

T1m3rlan, 2025-01-01:
v 2.2.0
Same on A1 mini

datacoda, 2025-03-17:
I'm not sure why it the link asks for elevated permission to register the DLL, but it should be the current user.  Try fixing it by opening up Command Prompt, go to C:\Users\_USERNAME_\AppData\Roaming\OrcaSlicer\plugins and run "regsvr32 BambuSource.dll" as the same user you're logged in as.

Makavel42, 2025-03-27:
datacoda solution worked for me with the latest version (2.3.0). 

Since my account doesn't have admin rights, the command was run on the admin account, not mine.

If you want things to be slightly easier:

1) Open the command prompt (either run or search for cmd.exe)
2) Enter the following line, no need to modify anything : regsvr32 %appdata%\OrcaSlicer\plugins\BambuSource.dll

closed 2025-07-03 by github-actions[bot]