OrcaSlicer/OrcaSlicer #5756: Multicolor not slicing correctly

adolfotregosa, 2024-06-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0 - 2.1 rc

### Operating System (OS)

Linux

### OS Version

manjaro

### Printer

voron 2.5

### How to reproduce

open stl, paint something , slice

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[Voron_Design_Cube_v8.zip](https://github.com/user-attachments/files/15892838/Voron_Design_Cube_v8.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2024-06-20:
Set shrinkage to 100% for your filaments. It will slice correctly then. Multicolour printing and shrinkage compensation are not compatible with each other at this point in time.

[image]

igiannakas, 2024-06-20:
Please test this fix here: https://github.com/SoftFever/OrcaSlicer/pull/5774

BrinksDaniel, 2024-06-22:
Having same issue, shrinkage already is 100%

igiannakas, 2024-06-23:
@BrinksDaniel please post your project file. Also which OS are you running, it’s version and what computer it’s on (Mac M series, Mac intel, PC).

igiannakas, 2024-06-25:
Latest Mac build should solve the non shrinkage related MMU issue. Download from the nightly and use this command to "sign" it so it opens.

`xattr -dr com.apple.quarantine /Applications/OrcaSlicer.app`

closed 2024-08-28 by SoftFever