OrcaSlicer/OrcaSlicer #5758: Add Option To Select Which Changes To Be Saved

cubitik, 2024-06-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be great if it was just like PrusaSlicer which gives us options to select what changes we want to save by ticking and unticking.

cubitik, 2024-06-18:
[image]

tlhintoq, 2024-06-19:
Merge with https://github.com/SoftFever/OrcaSlicer/issues/5749 ??

cubitik, 2024-06-19:
> Merge with https://github.com/SoftFever/OrcaSlicer/issues/5749 ??

Similar but not the same, I think this is better. Or both could be added but this is still my side.

closed 2024-09-25 by github-actions[bot]