OrcaSlicer/OrcaSlicer #5762: Printables.com support/import opens a new Orca instance for each file - Import to open instance

PhilBaz, 2024-06-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1 RC

### Operating System (OS)

Windows

### OS Version

windows 10

### Additional system information

Nothing important.

### Printer

Voron 2.4

### How to reproduce

Go to printables
use open in Prusaslicer button next to a file
repeat

### Actual results

Each time 'open in Prusaslicer' button is used a new separate instance of Orcaslicer is opened with just the one file.

### Expected results

Each time 'open in Prusaslicer' button is used 

1) If Orca slicer is **not** open it should open Orca slicer
2) If one instance of Orca slicer **is** open it should open the file in the running instance of Orca slicer.
3 If more than one instance of Orca slicer is open it should open in the last used instance of Orca slicer.

### Project file & Debug log uploads

[Q-2585 (Order).zip](https://github.com/user-attachments/files/15895057/Q-2585.Order.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

foreachthing, 2024-06-19:
You'll need to enable the option "allow only one OrcaSlicer instance":

[image]

> 'open in Prusaslicer' ❓❓❓

olcayoren, 2024-06-19:
This feature does not work on macos. Is there a way to adjust this?

olcayoren, 2024-06-19:
> You'll need to enable the option "allow only one OrcaSlicer instance": [...]

This feature does not work on macos. Is there a way to adjust this?

PhilBaz, 2024-06-19:
@foreachthing Okay, good to know.... But I like to have multiple instances of Orca slicer open. For different machines and materials etc. 

It would be nice if this feature was somehow compatible with multiple instances of Orca.

PhilBaz, 2024-06-19:
Maybe there can be a use one instance of Orca for Import option where it will import to the same instance as long as there is only one instance open? Any hacky solution would be good. 

It would just be too bad to have to choose between imports and multiple instances.

foreachthing, 2024-06-20:
@PhilBaz, well, I don't like "hacky solutions" to a problem that really isn't one. Just _import_ [CTRL+I] your 3mf to begin with - don't double click it.  That's just my opinion... 😉

PhilBaz, 2024-06-23:
@foreachthing This is only with regard to the printables import button, where you click the button right on the webpage. Not for files on the PC. I think the [CTRL+I] vs double click if for files on the PC.

It really will be a shame if you have to choose between printables import working in a way that's helpful, vs only ever having one instance of Orca open. For those of us with multiple printers having more than one instance open is critical. But then the way its currently implemented that means the printables import function will not really be helpful since each button push will launch a new instance of Orca.

closed 2024-09-29 by github-actions[bot]