OrcaSlicer/OrcaSlicer #5765: Preview image is confused about colors

Arakon, 2024-06-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0-rc

### Operating System (OS)

Windows

### OS Version

Win 11

### Printer

Bambu A1 Mini

### How to reproduce

- Prepare to slice a 2 color image with several objects, all in the same color except for a single colored text on the first layer
- In the preview window, several parts will be colored in one, and others in the other color
- The sliced preview and the preview image sent to the printer shows it correctly

### Actual results

Preview shows objects in different colors than actually selected
[image]
[image]

### Expected results

Preview should show the correct colors, otherwise very confusing. Also, this seemed to be okay in 2.0 and possibly 2.1 beta.

### Project file & Debug log uploads

[confused_colors.zip](https://github.com/user-attachments/files/15900004/confused_colors.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

frasier2k, 2024-06-29:
I can confirm this. I have similar behaviour with Orca Slicer 2.1.1

[image]

Linnie73, 2024-07-01:
I agree, noticed the same, Orcaslicer 2.1.1. Using AMS with multiple colors loaded. The preview window (shown when sending the print to the printer) initially shows the correct color but after a second or so, switches to the color of the next filament in line. Unless it is the last filament, then it remains the correct color. It seems to work properly on multicolored prints, this only happens on single color prints while using the AMS.

sschlesier, 2024-07-06:
I've experienced the same issue with a 2 color print where I print by object.  The Orca Slicer preview showed the wrong colours.   After sending to printer the embedded icon and the print itself used the correct colours.  Can work around this, but it is certainly confusing.

[image]

TinyMito, 2024-08-05:
I just installed 2.2.0-dev (nightly build), it seems to be fixed in that version.

brjohnso1, 2024-08-15:
I'm running 2.2 dev and it's still broken.

Goodwine, 2024-10-01:
I have a hunch it was introduced in this commit:
https://github.com/SoftFever/OrcaSlicer/commit/27f140fb1889ab6890aafd6c0035235cf5316772

closed 2025-01-31 by github-actions[bot]