OrcaSlicer/OrcaSlicer #5768: Messages blocked from veiw

thisnyjared, 2024-06-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Ender 3

### How to reproduce

On a smaller screen no matter what you do the gcode bar always blocks the message.
[image]

### Actual results

I can't see the message or act based on what it says
[image]

### Expected results

I should be able to see the message and click links I think it is because I have a small laptop screen with a resolution of 1280 x 720

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

It happens reguardless of any thing I do including the project or even clicking the button there.
[image]
[image]

foreachthing, 2024-06-20:
Yeah, I hate that, too! Just press [C] on your keyboard to hide the gcode window.
I wish for the notification to be on top of the gcode window.  :crossed_fingers:

shashachu, 2024-07-21:
> Yeah, I hate that, too! Just press [C] on your keyboard to hide the gcode window. I wish for the notification to be on top of the gcode window. 🤞

Thank you for this tip! I'm a new Orca user from Bambu Studio and I could not figure out how to hide that window.

closed 2024-10-28 by github-actions[bot]