OrcaSlicer/OrcaSlicer #5769: Weird wall route that leads to issues

abrilevskiy, 2024-06-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5 (23F79)

### Printer

Ultimaker 2+

### How to reproduce

Slice attached file [Cover.3mf.zip](https://github.com/user-attachments/files/15904062/Cover.3mf.zip)

### Actual results

[image]
A lot of isolated disconnected walls that leads to issues

### Expected results

The grid pattern is recognized and printed as lines

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15904117/log.zip)
[Cover.dbg.3mf.zip](https://github.com/user-attachments/files/15904118/Cover.dbg.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

discip, 2024-06-22:
Try this:
Just rename by deleting **`.txt`**.
[Cover.dbg.3mf.txt](https://github.com/user-attachments/files/15938325/Cover.dbg.3mf.txt)

abrilevskiy, 2024-06-25:
> Try this: Just rename by deleting **`.txt`**. [Cover.dbg.3mf.txt](https://github.com/user-attachments/files/15938325/Cover.dbg.3mf.txt)

It does the trick, but it is barely possible to define precisely the size of the grid. Thank you for the workaround, but it would be nice to have a fix since it might affect at different cases.

discip, 2024-06-25:
This should be almost identical to your original file.
[Cover.dbg.3mf.txt](https://github.com/user-attachments/files/15980011/Cover.dbg.3mf.txt)

closed 2024-10-01 by github-actions[bot]