OrcaSlicer/OrcaSlicer #5770: STL MESH ERROR

Frank-EBLUE, 2024-06-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

v2.1 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Device name	DESKTOP-O75NO12
Processor	Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz   1.80 GHz
Installed RAM	8,00 GB
Device ID	BA3539D8-2031-4755-A5D0-E98A5DA4F02C
Product ID	00342-23706-60015-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and single touch support

### Printer

ITOPI REPRAP CREALITY 4.27 board

### How to reproduce

1: STL Loaded into Orca
2: Sliced
3: sent to printer

### Actual results

Gcode and file preview is correct however the printed part prints as if it has mesh errors. This happend to a few diferent STLs i uploaded.  This specific one, part on the right fails whilst the one on the left prints perfectly.
[image]
[image]

### Expected results

Both files should print the same. Did a test on a another slicer and came out perfect. 

### Project file & Debug log uploads

[Uploading ORCA-MESH-BUG.zip…]()

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

I had around 4 STLs with the same type of issue regarding the mesh.

discip, 2024-06-19:
Have you by any chance the **`Arc fitting`** feature enabeled?

[image]

If so, your machine is most likely not able to process **`G2 / G3`** commands.

Please try again without that box checked.

discip, 2024-06-19:
@Frank-EBLUE
Have you by any chance the **`Arc fitting`** feature enabeled?

[image]

If so, your machine is most likely not able to process **`G2 / G3`** commands.

Please try again without that box checked.

discip, 2024-06-19:
@Frank-EBLUE
Have you by any chance the **`Arc fitting`** feature enabled?

[image]

If so, your machine is most likely not able to process **`G2 / G3`** commands.

Please try again without that box checked.

closed 2024-09-25 by github-actions[bot]