OrcaSlicer/OrcaSlicer #5776: Fast  fan speed change  on cooling overhang

pterodaktil02, 2024-06-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

If there are small overhangs, the program adds many short changes to the fan speed. 
This makes no physical sense: the fan will not be able to accelerate in hundredths of a second

Moreover, an error occurs on working equipment https://klipper.discourse.group/t/mcu-sb2040-shutdown-timer-too-close/16413/366

[Untitled.zip](https://github.com/user-attachments/files/15916872/Untitled.zip) 

Fill the table with this model and slice with the Force Cooling overhangs and bridges option enabled

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Add an option to ignore small changes in fan speed.

TheUnlimited64, 2024-06-21:
Maybe the Spinup time fixes that? In the printer settings you can set 2 settings regarding the fan. I dont know the english terms because my Orcaslicer is german, but try searching for "fan spinup time" or something like that. Its in the general printer settings (atleast for bambulabs)

pterodaktil02, 2024-06-22:
> Maybe the Spinup time fixes that? In the printer settings you can set 2 settings regarding the fan. I dont know the english terms because my Orcaslicer is german, but try searching for "fan spinup time" or something like that. Its in the general printer settings (atleast for bambulabs)

1. This makes no physical sense - the fan will not have time to change the speed and air flow in a tenth of a second
2. It creates redundant and dirty G-code
3. This leads to a DOS attack on the printer hardware: the printer is forced to process many meaningless commands. 
4. In some cases, this may cause printing to fail.

TheUnlimited64, 2024-06-22:
?
The 2 settings target the issue that the fan has no infinite acceleration. If you set spin up time to 1s, I believe it turns it on 1s before it's needed.

That would potentially fix the problem, but I'm with you, that sections with 0.1s fan of, Should be ignored for example 

But try it first, maybe that fixes your issue entirely

pterodaktil02, 2024-06-22:
In principle, it is possible to solve the problem at the printer level, but this is not correct. 
Because  the printer must honestly process the commands it receives. Otherwise it's a bad printer. 
The task of making the right set of commands is the task of the slicer.

TheUnlimited64, 2024-06-22:
I think we're talking past each other, in orca slicer you can tell the slicer, that the printer has not infinite spin up time, that way the slicer knows that it needs to spin up the fan beforehand 

So the printer itself has nothing to do, just execute gcode as normal and the fan gcode command is just a few commands beforehand

pterodaktil02, 2024-06-22:
Apparently I don't know Orca slicer that well. Tell me where to adjust the fan spin-up time in Orca

TheUnlimited64, 2024-06-22:
[image]
Press there

[image]
Theres the settings, i didnt tempered with it yet, but i believe it would fix it

pterodaktil02, 2024-06-22:
[image]
That  was a problem -  in  my  model  was  long  overhang - small no overhang  - small   overhang  - small no overhang -     long  overhang 
and   it  give    picture  like this 
[image]
Problem solved  -remove  checkbox   and  all  become  fine

closed 2024-06-22 by pterodaktil02