OrcaSlicer/OrcaSlicer #5777: Creality Ender-3 v3 KE 0.6 nozzle profile

marinegeek23, 2024-06-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

there's no profile for that printer with a 0.6 nozzle

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Please add 0.2, 0.6, 0.8 nozzle profiles for Ender 3 v3 KE

### Describe alternatives you've considered

I've tried modifying 0.4 and creating my own and it did not work out well

drawgas, 2024-07-05:
Would need this as well

Ramkethar, 2024-11-06:
Did anyone find the profile? I am trying the K1 profile 0.6 in orca

drewscott, 2024-12-26:
This would be great

DarkFisk, 2025-02-03:
I created myself, copied from 0.4, only  changed on Extruder tab option Nozzle diameter to 0.6, works. but better to have it from the box..

rAdziuB, 2025-02-07:
I also requested more nozzle sizes. This will be invaluable since my settings are still not perfect.

drawgas, 2025-02-07:
Good catch, mistyped. Should be height
> @drewscott how is possible to print line width 0.3 using nozzle 0.6?

dingausmwald, 2025-02-24:
Why set the first layer to 0.5 and others to 0.6 when for the 0.4 profile its the other way arround (bigger first layer, smaller - still bigger than nozzle - following layer)?

J3LS, 2025-02-25:
There is an Ender 3 v3 **SE** profile for a 0.6 nozzle. Does anyone know if we can use the printing parameters from that?

J3LS, 2025-02-25:
OK, so I have a unicorn 0.6 nozzle coming tomorrow. I'll set the layer widths and heights as above, do the calibration test prints and post back. If it works, looks like I can export the profile hopefully somewhere useful for folks.

kuzja111, 2025-09-21:
please add more nozzle size configurations: 0.6, 0.8, 0.2, 1.0

maor-avni-hpe, 2025-09-28:
#10860 adresses this.

kuzja111, 2025-09-28:
sorry, probably missed it, thanks!

thelegendtubaguy, 2026-11-08:
@marinegeek23 I believe this has been addressed.  If so, please close.