OrcaSlicer/OrcaSlicer #5780: Small area flow compensation - RC

adolfotregosa, 2024-06-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1-rc

### Operating System (OS)

Linux

### OS Version

manjaro

### Printer

voron 2.4

### How to reproduce

[small area flow compensation test.zip](https://github.com/user-attachments/files/15919077/small.area.flow.compensation.test.zip)
Rotate Z 270º 

### Actual results

[image]

2.1 - rc

### Expected results

[image]

2.1 beta

### Project file & Debug log uploads

[small area flow compensation test.zip](https://github.com/user-attachments/files/15919113/small.area.flow.compensation.test.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

not working on RC

adolfotregosa, 2024-06-21:
Not working on stable 2.1

Am I missing something ???

[image]

drewmoseley, 2024-06-21:
Same behavior here.  I tested on the 2.1.0 release on both Ubuntu (AppImage) and Mac (arm64/DMG).

mfish38, 2024-06-22:
Not working for me either. 2.1.0 release on Kubuntu AppImage

Screenshot taken with small area flow compensation on.
[image]

mjonuschat, 2024-06-22:
Looks like it was broken by https://github.com/SoftFever/OrcaSlicer/commit/9a4429f6037bb87001b78d9fe9f158bd5f1f4a61

AlexDF99, 2024-06-23:
I have the same problem, (2.1.0 and 2.0.0 final release) on Windows 11

TheSlashEffect, 2024-06-24:
The issue persists on Linux as well.
In its current state, `m_config.small_area_infill_flow_compensation.value` is not read as true when the feature is enabled, making the feature not activate.
If I revert https://github.com/SoftFever/OrcaSlicer/commit/9a4429f6037bb87001b78d9fe9f158bd5f1f4a61 the feature works as expected again.
Alternatively, if I move the definition of the config value small_area_infill_flow_compensation from PrintRegionConfig to PrintConfig, the feature works again.
The value was moved to the region config due to [this conversation](https://github.com/SoftFever/OrcaSlicer/pull/3334/commits/44665b39e1a43c29fa2bf9a4d566c89838545969#r1459245343):
I tried making the feature work again in its current state but to no avail. I ported the feature with the same PrintConfig setup on PrusaSlicer and SuperSlicer and it works as expected in both cases.

SoftFever, 2024-06-24:
Thanks, I will take a look

closed 2024-06-24 by SoftFever

SoftFever, 2024-06-24:
This should be fixed now.

PR #5844 (2024-06-24): Fix small area flow comp regression and some other small tweaks
# Description

- Fix regression, always create SmallAreaInfillFlowCompensator model
  `small_area_infill_flow_compensation` is per-region parameter, it's not necessary to iterate though all regions to check whether to create a model.

- Make SmallAreaInfillFlowCompensator more robust
  1. handle spaces/tabs/new lines etc.
  2. don't throw exception, fall back to no-op instead if parsing failed

- Fixing an issue that changing small_area_infill_flow_compensation per modifier didn't take effect

Fixed #5780