OrcaSlicer/OrcaSlicer #5788: Slow down for curled perimeter don't use Bridge speed

tome9111991, 2024-06-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1 RC

### Operating System (OS)

Windows

### OS Version

10

### Printer

Vron 2.4

### How to reproduce

Slice with Slow down for curled perimeter and without Slow down for curled perimeter.

### Actual results

Slow down for curled perimeter is override Briding speed for external briding.
I think Slow down for curled perimeter is dedect a Bridge as 100% overhang an use the overhang speed 75% - 100%

as an example
i have set overhang 75%-100% to 10mm/s
Bridge external to 50mm/s

a Bridge permimeter will move with 10mm/s

### Expected results

Slow down for curled perimeter  use the Briding speed for external briding.

maybe easy to fix if you set threshold for overhangs 75%-99% ?
and briding external use 100% threshold?

### Project file & Debug log uploads

[slow down.zip](https://github.com/user-attachments/files/15926258/slow.down.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-06-21:
That is intentional - the overhang perimeter is not technically a bridge but rather a perimeter with 100% overhang. 

Unfortunately the code cannot distinguish between a perimeter that happens to be 100 overhanging and an overhang perimeter that is part of a bridge. 

So we conservatively use the overhang perimeter speed. Else you'd have the scenario where in external perimeters that are overhanging the speed would be going up to the bridge speed instead of down to the 100% overhang perimeter speed.

tome9111991, 2024-06-23:
ok I'm not sure why.
but internal bridge is using the speed for internal bridge works so far.
[image]

i dont know why not work for external bidge.
here you can see for external bride it will use the speed for 75%, 100% overhang
[image]

igiannakas, 2024-06-23:
As I said above it’s intentional… for the cases where the external perimeter is overhanging fully but is not a bridge. If you want to use the faster speed you can disable slowdown for curled perimeters or simply increase the 75-100 speed. 

However in any case you’re going a bit slow for the overhangs - you shouldn’t need to slow down to 15 and 10 mm/sec. I have set these to 0-60-40-30 in my profiles. And the bridge speed I’ve also set it to 30 for the external bridges. That combination works extremely well ;)

closed 2024-06-23 by tome9111991

tome9111991, 2024-08-19:
> Unfortunately the code cannot distinguish between a perimeter that happens to be 100 overhanging and an overhang perimeter that is part of a bridge.
> 

but the slicer can already distinguish between overhang and briding.

the light blue is bride perimeter but is printed with overhang speed instead of Bridge speed
[image]

igiannakas, 2024-08-19:
That is not a perimeter. When printing a perimeter there is no distinction between overhang and bridge. Check the gcode preview and you’ll see that the perimeters attached to a bridge are tagged as overhang perimeters ;)

tome9111991, 2024-08-19:
i tried the latest nightly, looks fixed.
THX

igiannakas, 2024-08-20:
> i tried the latest nightly, looks fixed. THX

Perfect :)