OrcaSlicer/OrcaSlicer #5794: Treat high/low filaments combination as warning but not error

vovodroid, 2024-06-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently you can't combine high/low temperature filaments as support interface material due to error

[image]

In reality there is no problem whatsoever to use them together, PLA temperatue could be about 210-220, ABS/ASA 220-240. So such formal restriction is too strict. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Just give warning,  or better use actual printing temperature difference from filament properties for such warning, but don't forbid slicing and let to user to decide.

Nathan22211, 2024-06-23:
this would def be a step into getting toolchanger support into Orca. I had to trick orca by putting the filament temps in the filament start gcode with my SV04 whenever I wanted to do a PLA/ABS print previously.

vovodroid, 2024-09-22:
ping

closed 2024-12-29 by github-actions[bot]

zbrozek, 2025-04-05:
I see this is closed as not-planned. Could it please be planned? I don't want the slicer trying to protect my from myself.