OrcaSlicer/OrcaSlicer #5796: Add custom Prepare time for Klipper

O-to-the-o, 2024-06-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Total time estimation of by the time my printer takes to prepare 

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

In the Total estimation there is the Prepare Time, it seems to be dependend on the starting g-code. In klipper many moved that to the config on the klipper host. So the Start-Gcode is just Print_Start + Parameters and the estimation gives always 1s.

I'd like to get a Parameter in the Printer Settings that just sets the Prepare time to a custom value and adjust the Total Time accordingly. This would improve the estimate for me and others

### Additional context

[image]

damienvdw, 2024-07-17:
Honestly, this could be good for all printers !

I'm running my BBL P1S with custom G-code and the inaccurate prep time is bothering me so much...

A "Prepare time override" setting in the printer profile would be amazing.

HugoCLSC, 2024-08-21:
Yes! the possibility for adjusting this paremeter, in my opinion, whould further increase the accuraty of the print time estimate.

As a bonus, QoL update, because we'll be able to just see how much time it takes for the printer to actually start printing.

Adjusting this parameter, whould in fact be amazing. For every printer that uses Orca

freiser77, 2024-11-01:
Yes, it would be beneficial. I have a printer profile for big parts where the printer idles for 10 minutes to heat the bed evenly. I do that with a G4 command, but it is not considered in the preparation time. This would allow me to circumvent the time estimation problem.

Dnic94, 2024-12-23:
I also stumbled over this issue. Would also be interested in an enhancement.

MistressPlague, 2025-01-14:
I would also like this.

Minecreator200, 2025-02-05:
It would surely be a great enhancement. Especially when included in the presets.

chimcen, 2025-06-13:
This would be great - any news here?

BanzoV, 2025-08-02:
This would be great!

tealblu, 2025-08-09:
Yeah, I'd appreciate this as well!

NIKOSTERs, 2025-08-20:
This feature would be great! I can't rely on the estimate from the slicer because it takes my printer up to 20 minutes to heat up and calibrate the bed

Ken6432, 2025-08-24:
Perhaps the setting should be placed in the filament settings.  It seems like the prepare time is driven mostly by the time it takes for the bed and chamber to heat up which is more filament driven than machine driven.

broncosis, 2025-08-30:
I'll also say that this would be a nice Quality of life improvment

AndyHazz, 2025-09-16:
Or, as it's designed to look at the start g-code and work out prepare time from that, allow a special comment in that existing field (something obvious like `; Prepare time = 300`) to allow the prepare time to be overidden ...

mcollins1290, 2025-10-24:
I want this as well! I know at least 4 or 5 of my friends who have asked me whether I knew whether there was a way to manually set this, this would be such a great feature!!!!

iinsaane, 2026-01-21:
I would also like this. Based on gcode, orca guesses 26s, but total is ~6:30 with the sv08 quad gantry calibration and bed mesh

myersvx, 2026-02-19:
I too would like this as a feature! Just a spot to add prep time. Preferably to filament. but if not then to printer. even if only, as previously suggested, as a value to set in the start gcode section.