OrcaSlicer/OrcaSlicer #5802: Setup Wizard does not load on Arch

CitrusIntellect, 2024-06-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Additional system information

OS: Arch Linux
KERNEL: 6.9.5-arch1-1
CPU: 13th Gen Intel Core i7-13700F
GPU: NVIDIA GeForce RTX 4060
GPU DRIVER: NVIDIA 550.90.07
RAM: 32 GB
DE: Kde Plasna 6.1.0 (Wayland)

### Printer

Software Side

### How to reproduce

Use appimage in terminal to open OrcaSlicer, nothing loads

### Actual results

Nothing Loads

### Expected results

Setup Manager should come up so I can do setup.

### Project file & Debug log uploads

LOG:
----START OF LOG----
[2024-06-22 00:02:16.492588] [0x000079e7eca9cb80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:20391): GLib-GObject-CRITICAL **: 00:02:17.232: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:20391): Gtk-CRITICAL **: 00:02:17.232: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:20391): GLib-GObject-CRITICAL **: 00:02:17.233: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:20391): Gtk-CRITICAL **: 00:02:17.233: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:20391): GLib-GObject-CRITICAL **: 00:02:17.235: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:20391): Gtk-CRITICAL **: 00:02:17.235: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:20391): GLib-GObject-CRITICAL **: 00:02:17.236: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:20391): Gtk-CRITICAL **: 00:02:17.236: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:20391): GLib-GObject-CRITICAL **: 00:02:17.237: invalid cast from 'wxPizza' to 'GtkCellLayout'

[... 278 lines cut ...]
(orca-slicer:20391): GLib-GObject-CRITICAL **: 00:02:21.971: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:20391): Gtk-CRITICAL **: 00:02:21.971: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:20391): GLib-GObject-CRITICAL **: 00:02:21.976: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:20391): Gtk-CRITICAL **: 00:02:21.976: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:20391): GLib-GObject-CRITICAL **: 00:02:21.976: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:20391): Gtk-CRITICAL **: 00:02:21.976: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (orca-slicer:20391): CRITICAL **: 00:02:22.011: Cannot register URI scheme wxfs more than once

** (orca-slicer:20391): CRITICAL **: 00:02:22.011: Cannot register URI scheme memory more than once
Failed to get GBM device
Failed to get GBM device
Failed to get GBM device
Failed to get GBM device
----END OF LOG----

### Checklist of files to include

- [X] Log file
- [ ] Project file

darrellenns, 2024-06-27:
I believe this is an bug with webkit2gtk and nvidia drivers. You can work around it by setting the environment variable WEBKIT_DISABLE_DMABUF_RENDERER=1.

Similar issues are reported in multiple places. For example:
https://gitlab.archlinux.org/archlinux/packaging/packages/webkit2gtk/-/issues/1
https://github.com/tauri-apps/tauri/issues/9304
https://bugs.archlinux.org/task/79783 (same issue happened with an older webkit2gtk version)
https://bugs.webkit.org/show_bug.cgi?id=261874 (same issue happened with an older webkit2gtk version)

CitrusIntellect, 2024-07-28:
That did the trick! Thanks

closed 2024-07-28 by CitrusIntellect