OrcaSlicer/OrcaSlicer #5803: Mintion beagle v2 crashes when using orcaslicer gcode but not cura

bushibot, 2024-06-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

Modern i9 game box, likely not relevant.

### Printer

Anycubic Vyper, Marlin 2.0.9.2

### How to reproduce

Mintion beagle is camera has a pi like board running something like octo print. You upload files over wifi and it sends to printer.
1. Slice file with Orcaslicer
2. upload to cam
3. start print
4. extruder will move around a bit (seems random but not liking something in the start gcode I guess).
5. extruder and might or might not start to heat.
6. Process hangs and never continues.

I can slice the same file with Cura and it starts printing with no issue. Seems tied to something in orca header.
I opnend a support case with the company but have not seen anything useful come back. 

### Actual results

Extruder move around a bit (it shouldn't) ending over bed, then starts bed/extruder heating (inconsistent). After heating nothing else happens, print does not start.

### Expected results

Print to start normally like it does with a cura sliced file. Honestly it's probable bad paring on their end, but reporting it here anyway. Happy to try any suggestions. 

### Project file & Debug log uploads

NA can reproduce with any stl.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

bushibot, 2024-06-22:
[6-32_standoff_motherboard_mounting_plate_v1.4_PETG_7h38m.txt](https://github.com/user-attachments/files/15935689/6-32_standoff_motherboard_mounting_plate_v1.4_PETG_7h38m.txt)
attaching gcode as txt.

bushibot, 2024-06-22:
Looks like it's something in this block that is confusing the beagle
```
; EXECUTABLE_BLOCK_START
M73 P0 R458
M201 X1000 Y1000 Z200 E5000
M203 X200 Y200 Z10 E60
M204 P1250 R1250 T1250
M205 X8.00 Y8.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
M190 S80 ; set bed temperature and wait for it to be reached
M104 S225 ; set nozzle temperature
;TYPE:Custom
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G0 Z0.2 F1800 ; move nozzle to print position
G92 E0 ; specify current extruder position as zero
G1 Y10 X180 E50 F1200 ; extrude a line in front of the printer
G92 E0 ; specify current extruder position as zero
G0 Z20 F6000 ; move head up
G1 E-7 F2400 ; retract
G04 S2 ; wait 2s
G0 X0 F6000 ; wipe from oozed filament
G1 E-1 F2400 ; undo some of the retraction to avoid oozing
G1 F6000 ; set travel speed to move to start printing point
M117
G90
G21
M83 ; use relative distances for extrusion
; filament start gcode
M900 K0.02; Override pressure advance value
M106 S0
```

bushibot, 2024-06-22:
Manted to catch the console error this time when I stated the oraca print
```
ecv: T:96.67 /225.00 B:80.01 /80.00 @:127 B@:127
Send: N45 G1 X9.964 Y232.271 E.04615*96
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: ok
Send: M105
Recv: echo: cold extrusion prevented
Recv: ok
Printing is stoping!PrStater=102,RspStater=4
Send: M114
Send: M108
Recv: Error:Line Number is not Last Line Number+1, Last Line: 45
Recv: Resend: 46
Printer had stoped!
Send: M105
```
So on my gcode with the header block removed I think that would be
```
;_SET_FAN_SPEED_CHANGING_LAYER
M204 S1000
G1 Z.4 F7200
G1 X14.005 Y16.824
```

BeagleCamera, 2024-06-26:
Hello
First of all, thank you for your detailed feedback. Based on our investigation and verification, the error feedback indicates "hot end temperature is too low."

After careful inspection, we found that the Vyper configuration in OrcaSlicer causes the printer to start loading or unloading filament without waiting for the hot end to reach the target temperature.↳

Please refer to the attached figure "Temperature Setting." When we manually modified the Gcode file or adjusted the OrcaSlicer printer configuration (adding a command to wait for the hot end to reach the target temperature), the printing proceeded normally. 
Refer to the attached pictures "Manual modification" and "Slicer software modification."↳

Therefore, the issue you reported is mainly due to a bug in the OrcaSlicer Vyper configuration file. Additionally, there is room for optimization in the error feedback provided by the Camera.↳

[image]
[image]
[image]
[image]

bushibot, 2024-06-26:
So what’s interesting is it prints fine through octo print and on the printer directly without issue. I raised a gut issue with orcaslicer… I’m just a user I suggest posting there. As is the products are not compatible, which is shame since orca is arguable the best open source slicer out.-Griffon

1.jpg (view on web)
2.jpg (view on web)
3.jpg (view on web)
4.jpg (view on web)

BeagleCamera, 2024-06-27:
In the terminal of Octopi, you can also see the error message about the extruder being too cold, but it does not stop printing. However, the BeagleCamera stops printing when this error occurs.
Please recognize the BeagleCamera's cautious approach, which helps prevent further damage to the printer.
[image]

bushibot, 2024-06-30:
The heart of the issue seems to be that Orcaslicer is using a M104 instead of M109, witch beagle seems to not like and. Octo and the printer don't seem to parse the difference the same way and wait just like if it were a M109. If a manually change the gcode and replace the M104 with M109 and the start, then the beagle streams it normally without erroring out. Beagle support argues that their error handling is correct and properly more conservative. I'm not knowledgeable enough to argue, I can only just point out the difference in behavior as observed.
Maybe a somebody on the Oracaslicer team could comment on why the M104 at start.
```
; EXECUTABLE_BLOCK_START
M73 P0 R21
M201 X1000 Y1000 Z200 E5000
M203 X200 Y200 Z10 E60
M204 P1250 R1250 T1250
M205 X8.00 Y8.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec
M190 S80 ; set bed temperature and wait for it to be reached
M104 S225 ; set nozzle temperature
```

bushibot, 2024-07-18:
Not sure why this was closed? It's legitimate orca issue.

closed 2024-10-25 by github-actions[bot]

bushibot, 2025-04-21:
A year later Orca gcode still dose not right, still uses 104 instead of 109, which is improper for printer start temps