OrcaSlicer/OrcaSlicer #5808: k1c nozzle profile dont show up.

CandWo, 2024-06-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

k1c

### How to reproduce

add printer k1c
nozzle 0.6,0.8

### Actual results

no printig profile . default more not
[image]

### Expected results

by the k1 ist comming

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15936929/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

i had crashes but dont no why . its chainese for me

BruffyT, 2024-06-22:
Ive taken a look through the config files and can see that no one has created 0.6mm or 0.8mm profiles yet for the K1/K1c/K1 max. 

I will take a look and try to create some for the K1c and do some testing on them.

BruffyT, 2024-06-22:
I believe i have a fix for this, i am just testing now :)

CandWo, 2024-06-22:
I saw it by all the k1 k1c k1max all have 0.6 0.4 0.8 nozzles by add a systemprinter . The k1 had a finish 0.6 profile 3-4 prcesprofiles variants of layerhight  and in the config i see all by creality and when i would manual add a new nozzle i cant make it i cant chose a printer

BruffyT, 2024-06-23:
Hey @CandWo the PR is now open for the fix and i have tested and all looks good, just waiting for a review/approval on it :)

CandWo, 2024-06-23:
@BruffyT  thanks . Very quick. I write it in the . Jons file and its showing up . Ive a question can you look in my log for the crashes i had? This crashes comin with beta 2.1 befor ive,nt this

BruffyT, 2024-06-23:
If I get chance to some evening sure.

BruffyT, 2024-06-24:
Were these crashes when you were trying to load the 0.6mm or 0.8mm nozzle profile?

CandWo, 2024-06-24:
No i can‘t reproduce it . Its random . Its coming when i make other and the slicer is open ,too.  Or when i have two orca slicer open or when i working on the models

BruffyT, 2024-06-24:
The only errors I can see in them logs are for file not found (cos of the bug which I fixed) but this shouldn't cause it to crash. It is a possibility windows is closing the app to cut down on resources if the cpu or memory is at 100% since it's only when doing other things as well.

BruffyT, 2024-06-26:
Hmm that's unlikely to be the problem then, if it persists and you can recreate it please raise another issue.

PR #5815 (2024-06-26): Resolves #5808: Fix missing K1C processes
Resolves #5808

Screenshots of evidance that they are now showing attatched below

Before not showing layer height profiles: 

[image]

After now showing the profiles:

**0.6mm Nozzle:**

[image]

**0.8mm Nozzle:**

[image]

Update after [ee52ec3]:

**0.6mm Nozzle:**

[image]

**0.8mm Nozzle:**

[image]
commit aeac07b9 (2024-06-26): Resolves #5808: Fix missing K1C processes (#5815)

* refactored printer name to resolve #5808

* updarted name of profile to match all other K1C profiles

---------
closed 2024-06-26 by SoftFever