OrcaSlicer/OrcaSlicer #5812: Ironing Inset

PeteCondliffe, 2024-06-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When ironing sharp edges are created and it also feels a little rough if you run your fingernail over them.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Implement a feature similar to what's available in cura to set an ironing inset so ironing doesn't go all the way to the edge.
Thanks

fbson, 2024-07-02:
I'd second the request please - sharp edges on ironed surfaces means a lot of part trimming for me at the moment so it would be most helpful I think.

idmadj, 2024-08-22:
This has been asked for in #3808 in the past (eventually closed by stalebot) as well as #2778, so there's definitely a significant amount of interest for this feature.

closed 2024-11-28 by github-actions[bot]

dfosantos, 2025-01-07:
https://github.com/SoftFever/OrcaSlicer/pull/7969