OrcaSlicer/OrcaSlicer #5813: Arrange all objects not working correct when printing by object

poppinfresh69, 2024-06-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1

### Operating System (OS)

Windows

### OS Version

OS 10

### Additional system information

Ryzen 5 2600x and GTX 1660TI

### Printer

A1 Mini and E3v2

### How to reproduce

1) load a small model and clone it
2) go to Special mode and enable print by object
3) notice collision warning and click Arrange All Objects
4) notice objects do not get moved apart to avoid collisions (A1 Mini profile and E3v2)
5) switch to P1S profile > click arrange and notice models move apart to avoid collision
6) switch back to A1 Mini profile > click arrange and notice models move from non-collision into collision state

### Actual results

Auto arrange behavior seems erratic depending on the printer profile when printing by object.

### Expected results

Arrange should move models out of collision automatically like it used to

### Project file & Debug log uploads

[Orca.zip](https://github.com/user-attachments/files/15938375/Orca.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]
[image]

regularlabs, 2024-07-04:
I can confirm this is still an issue in v2.1.1

theherox, 2024-08-14:
This was working for me in 2.0.0 but is broken in 2.1.1.

closed 2025-02-19 by github-actions[bot]