OrcaSlicer/OrcaSlicer #5816: Remove Associate web links to orcaslicer

bhast2, 2024-06-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

bambulab x1c

### How to reproduce

put check marks in the associate web links to orcaslicer. Can't remove the check marks

### Actual results

can't dissociate web links to orca slicer  

### Expected results

remove the check marks and not use weblinks anymore 

### Project file & Debug log uploads

[OrcaSlicer.zip](https://github.com/user-attachments/files/15942413/OrcaSlicer.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Deeply999, 2024-07-11:
I am using 2.1.1 and still an issue, unable to remove links associated to sites.

LampeMW, 2024-10-04:
Still unable to remove this. I wanted to test Bambu Studio for something and cannot open the links. Do I need to uninstall OrcaSlicer first?

closed 2025-01-09 by github-actions[bot]