OrcaSlicer/OrcaSlicer #5819: closed box slices open

bendelathouwer, 2024-06-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Win 10

### Printer

ender 3 v3 se

### How to reproduce

 slice a simple rectangular box 
[image]

### Actual results

[image]
 hollow box 

### Expected results

a solide box 

### Project file & Debug log uploads

[Desktop.zip](https://github.com/user-attachments/files/15943936/Desktop.zip)
[Desktop.zip](https://github.com/user-attachments/files/15943939/Desktop.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Arakon, 2024-06-23:
It's doing EXACTLY what you set it up to do.
No top surface, no infill. This is not a bug.
[image]

bendelathouwer, 2024-06-23:
Still never worked with a slicer that uses defaults like that , so might be worth changing that especially for the beginners  :)

closed 2024-06-23 by bendelathouwer