OrcaSlicer/OrcaSlicer #5821: Object not slicing correctly when painted

richard-scott, 2024-06-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

macOS

### OS Version

Sonoma 14.5

### Additional system information

I have a Mac mini M1 202.

### Printer

Bambu X1C with AMS

### How to reproduce

I have a medal made of two objects: the face plate and the outer ring. I selected different filaments for each object, and it looks okay in the slicer, but when you print it, the top layers all take the colour of the face plate and do not change for the outer ring. This only happens on the top layers; the bottom layers print fine.

### Actual results

In v2.1.0, I get this:

[image],

### Expected results

In v2.0.0, I get this:
[image]

### Project file & Debug log uploads

[Medal.3mf.zip](https://github.com/user-attachments/files/15944070/Medal.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2024-06-25:
Latest Mac build should solve the non shrinkage related MMU issue. Download from the nightly and use this command to "sign" it so it opens.

`xattr -dr com.apple.quarantine /Applications/OrcaSlicer.app`

richard-scott, 2024-06-26:
@ianalexis It's still the same in the latest build from [here](https://github.com/SoftFever/OrcaSlicer/actions/runs/9665273126).

richard-scott, 2024-06-26:
This is still the same in v2.1.1 but was working back in v2.0.0.

richard-scott, 2024-06-26:
The green centre and the orange outer ring are separate objects that are not painted but coloured by selecting different filaments from the AMS.

richard-scott, 2024-09-13:
This looks to be working now with [v2.2.0-beta](https://github.com/SoftFever/OrcaSlicer/actions/runs/10840202908).

closed 2024-09-13 by richard-scott