OrcaSlicer/OrcaSlicer #5824: after update to orca slicer 2.1 i cant see the printer tab in prepare so i cant change any settings for the slice

maxxx505, 2024-06-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

ender 3 v3 ke

### How to reproduce

i updated the slicear and then i cant see it

### Actual results

i cant slice anything with my settings

### Expected results

usinng slicer how i want

### Project file & Debug log uploads

??

### Checklist of files to include

- [X] Log file
- [ ] Project file

Avpman2, 2024-06-23:
I believe this happened to me, also. I installed the new version (prompted when the old program loaded), and all my printers and settings were gone when the new version ran. Reinstalling the previous version did not fix it. I had to restore my PC to the point before the update to the latest version. From now on, I will test the "portable" version so as not to muck up my installed version.

BrunoBrutus, 2024-06-25:
I'm only using the portable version, and I'm experiencing the same issue. I had three different printer profiles, but now the system prompts for a fresh start each time. 
First I added a new printer (BAMBU P1S) and removed the other selections. 
After restarting the program, I added generic KLIPPER and MARLIN 0.4 nozzle printers, and the other printers appeared. 
Then, I removed the Bambu, and the remaining printers persisted after restarting the program.

closed 2024-10-01 by github-actions[bot]