OrcaSlicer/OrcaSlicer #5829: 3mf file overwritten even without confirmation

manferlo81, 2024-06-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

Core i5, 16Gb ram, NVIDIA GeForce GTX 1650

### Printer

Ender 3

### How to reproduce

You have a folder with 2 files (model.step & model.3mf)... create a new project..., add model.step file... when you hit "save", the file is saved as model.3mf overwriting the original model.3mf without even a "do you want to overwrite the file?" warning.

### Actual results

3mf file overwritten

### Expected results

At least warn about it

### Project file & Debug log uploads

[model.zip](https://github.com/user-attachments/files/15944903/model.zip)

files are inside the archive, but you dont need them for this issue. there are 2 file model.step (containing a cube) & model.3mf (containing a cylinder)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I believe a distinction should be made about adding (or importing) a file and opening a project. When you have a new project and you add a model to it, the project sould not adopt the model filename, instead it sould keep its original filename "Untitled" or "New Project" or somethig like that. I understand sometimes it saves time to adopt the name but I believe it creates confusion. I think its better to let the user choose a filename when the project is being saved, and if the filename chosen already exists, OrcaSlicer should warn about it an let the user decide if overwrite or not.

discip, 2024-06-24:
2nd that!
This happened way to often!
A confirmation popup would be sufficient.

Akaitatsu, 2024-06-29:
Standard behavior for GUI applications would be to leave the filename as Untitled#.3mf until a Save As operation is performed. If the user clicks Save or hits Ctrl+S then it should do a Save As and prompt for a location and filename.

discip, 2024-09-28:
Still a very much needed feature.

discip, 2024-10-04:
@Noisyfox
Maybe you could look into this issue, Sir? 😊

Noisyfox, 2025-01-08:
I couldn't reproduce the issue. A save as dialog popped up as expected when I press the save button.

buzzhuzz, 2025-01-08:
I've seen that behavior before, but never payed attention to it.

I usually create new project, then import objects using "Add" button on the toolbar. After that upon pressing Ctrl+S  project getting saved into "model.3mf" beside original "model.stp" without asking where to save project to.

Noisyfox, 2025-01-08:
> Interesting, if you add the model via drag and drop like you did, it works, but if you add it using **`Ctrl+I`**, the existing file gets overwritten.

Ah you're right!

Noisyfox, 2025-01-08:
https://github.com/bambulab/BambuStudio/commit/0d8de21429209779ba81f6875622df9ad0628dda
https://github.com/bambulab/BambuStudio/commit/debbed13d6e2beb31e63299472e529ba70e6fce2

Noisyfox, 2025-01-08:
I think it's better to revert those "improvements" because:
1. it works differently than drag-n-drop, which is confusing
2. overwrite files sliently is WAAAAAY worse than not having a "smart project name based on imported file"

buzzhuzz, 2025-01-08:
I'm currently trying to make them as they should: set project name without assigning project 'file name', so first project save will result in "Save As" action.

Noisyfox, 2025-01-08:
With that said, I appreciate for your effort working on this issue, thanks!

closed 2025-01-08 by Noisyfox

manferlo81, 2025-02-07:
The greatest problem is not the filename, It's the fact that it writes the file without asking, even overwriting it if it already exists.

Imagine you already imported the `.step` file, apply custom supports, use paint on seams, apply some settings to some layers... or any other thing, you spent 10-20 minutes setting everything up perfectly, and save it as a project. Now by accident you import the `.step` file again and click save by accident again, now all your work from before its gone, without even asking. I hope everybody sees how serious this is. This issue should not be `closed`.

For now I'll try to import files using drag-n-drop, but IMHO `drag-n-drop` should work exactly the same as `import`. A distinction has to be made between "open" and "import".

manferlo81, 2025-02-28:
> > The greatest problem is not the filename, It's the fact that it writes the file without asking, even overwriting it if it already exists.
> > Imagine you already imported the `.step` file, apply custom supports, use paint on seams, apply some settings to some layers... or any other thing, you spent 10-20 minutes setting everything up perfectly, and save it as a project. Now by accident you import the `.step` file again and click save by accident again, now all your work from before its gone, without even asking. I hope everybody sees how serious this is. This issue should not be `closed`.
> > For now I'll try to import files using drag-n-drop, but IMHO `drag-n-drop` should work exactly the same as `import`. A distinction has to be made between "open" and "import".
> 
> Have you tried the latest dev build? Now it should pop up a save-as dialog instead.

I tried version 2.3.0-beta and it seems to be fixed, thanks!