OrcaSlicer/OrcaSlicer #5832: Additional / Different perimeter count on internal holes

Lazersfly, 2024-06-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am developing print settings for the lightest parts possible (for a 3d printed plane), and for many parts the primary contour (the "outer" surface) should be a single wall, but inner holes (such as for a bolt) would ideally be 2 or more walls for strength. 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add an option for extra perimeters on internal holes/contours. The X-Y hole compensation setting can identify and shrink/expand the targeted perimeters for this feature, so I imagine some of the code is already there for this feature.

### Additional context

[image]

Ideally the holes in this part would have two walls, where the outside would still only have one.

Lazersfly, 2024-06-23:
Additional context - I am aware of print modifiers, and use them extensively for adding/removing top surfaces and othehe and introduces some weird bugs (such as the infill sometimes generating holes at the seam between modifier/non-modifier areas).

henrivdr, 2024-07-18:
I absolutely upvote this one.

TylonHH, 2024-08-24:
https://github.com/SoftFever/OrcaSlicer/issues/6351

jzagaja, 2024-09-27:
Super Slicer already implemented this.

henrivdr, 2024-12-27:
This would be very useful for structural prints

Pbmarcsson, 2025-01-03:
Completely agree, just like you can choose the scarf joint on contour and or hole.

realhoolito, 2025-07-09:
upvote this! extremely useful to tap threads, leave thickness when increasing hole diameter by drilling, place inserts..... etc
You can already paint faces with the bucket for supports, so such could be the UI for specifying which holes

LupusTheCanine, 2026-01-19:
One thing to note that hole detection may yield unexpected results in what parts of the model are considered "holes", I would recommend using volume modifiers with 100% concentric infill set around the areas you want to reinforce.

discip, 2026-09-09:
That would be really neat feature. 😊