OrcaSlicer/OrcaSlicer #5836: Import Orientation Method from Prusa Slicer

cromine11, 2024-06-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The Arrange tool doesn't seem to be very effective. I have had incidents of overlap between parts and odd angles.
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

PrusaSlicer has already implemented a very good Arrange tool that puts things in a good orientation and never overlaps. I will typically arrange in PrusaSlicer just to have a good arrangement and then import to Orcaslicer. 
[image]

Additional suggestion:
Something that could also be added is print order organizer to optimize the order the parts are printed based off the arrangement so the tool head travels a reasonable distance to the next part or even do have it go to a part that is not right next to the one that was just printed so it has time to cool evenly. Give it a couple options.

discip, 2024-06-24:
This should be combined with #5830.

Arakon, 2024-06-25:
Seems to be the same issue as https://github.com/SoftFever/OrcaSlicer/issues/4719?

closed 2024-10-02 by github-actions[bot]