OrcaSlicer/OrcaSlicer #5837: layer 2 and 3 speed settings on the bridge

Br3n3n, 2024-06-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hello, I have a problem with the bridges. I would like to be able to adjust the speeds on the bridges. A layer that is already placed on the bridge layer so that the speed can be set. to make the final print better

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

by adding options to influence the speed of the layer, which can be the most, thus increasing the print quality (this is mainly high-speed printing)
this option is already in cura slicer

### Additional context

[image]

stanpete78, 2024-06-25:
+1 Very helpful setting!

digital0785, 2024-06-29:
this would be awesome!

also an option to go along with this is adding a bridge specific overlap setting so that way it can push further into the print 
this is just a temp tower but it basically just butts right up against it instead of overlapping so if you don't slow down enough it wont make good contact causing issues on subsequent layers :)

[image]

closed 2024-10-06 by github-actions[bot]