OrcaSlicer/OrcaSlicer #5840: Can't upload via PrusaLink anymore.

StayPirate, 2024-06-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Linux

### OS Version

Archlinux

### Printer

Prusa MK4

### How to reproduce

Slice -> Print -> "Upload" or "Upload and print"

### Actual results

Upload starts but it stucks at 0%, that can be seen both from OrcaSlicer and the MK4 display. After few seconds I get an error HTTP 408.

[image]

### Expected results

The upload should be successful.

### Project file & Debug log uploads

This is happening to me with any opened project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I dug into this and I find out the culprit commit: https://github.com/SoftFever/OrcaSlicer/commit/cd6cd0786fc494d2f1e2fd2123e791f65fbcc97a. So, I [reverted it](https://github.com/StayPirate/OrcaSlicer/commit/cbbd8d20b3b322d9fb4010a5b6e663f3da53c3fa) and rebuilt the AppImage, and the bug was gone. I tag @Noisyfox as he's the author of that commit. Could you please review it with this issue in mind?

Noisyfox, 2024-06-24:
I see, I think I've found the issue. Will raise a PR soon.

Noisyfox, 2024-06-24:
Could you give the PR a try (once it finishes building) and see if that solves the issue?

StayPirate, 2024-06-24:
Confirm, this PR fixed the issue on Linux too. Thank you very much 👍🏼

PR #5843 (2024-06-24): Fix PrusaLink upload when PUT is used
Fixes #5840