OrcaSlicer/OrcaSlicer #5841: Default part cooling fan speed

Javi3rV, 2024-06-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Maybe, it's so hard to control the part cooling fan speed rn

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be nice to have a switch "use gradual part cooling fan speed" and "use default part cooling fan speed".

Let me explain. Right now we have "min speed" and "max speed" at layer, that makes a gradual speed. But, what if we dont want a gradual fan speed? What if I want to set the fan speed at a continous XX%?

Arakon, 2024-06-24:
...then you simply set all values to 100% and it will always be at max speed.
[image]

Javi3rV, 2024-06-24:
> ...then you simply set all values to 100% and it will always be at max speed.
> [image]
> 

But that's not the result I'm trying to get, I mean, I want it always at 40% for example. That's impossible...

Arakon, 2024-06-24:
Then set all 3 values (min/max and overhangs) to 40% instead of 100%. The fan will remain at that speed at all times then.

Javi3rV, 2024-06-25:
> Then set all 3 values (min/max and overhangs) to 40% instead of 100%. The fan will remain at that speed at all times then.

That's still not an optimal solution. There is a lack of fan control anyway. What if I need the fan to stay at 40% but increase to 70% for overhang? There are a lot of situations that could be easier than setting everything to the same value

Arakon, 2024-06-25:
That's what the "Fan speed for overhangs" in my screenshot is for. I'm not sure what exactly you are trying to control that isn't already fully controllable right on that screen?

Javi3rV, 2024-06-25:
> That's what the "Fan speed for overhangs" in my screenshot is for. I'm not sure what exactly you are trying to control that isn't already fully controllable right on that screen?
> 

I think my question is easy to understand:
What if I need the fan to stay at continuous 40% but increase to 70% for overhang?

Just 2 values, always at 40% and 70% for overhangs.
Your solution would just keep the fan at 100%

Arakon, 2024-06-25:
No. Set min and max to 40% and overhangs to 70%. Always 40% UNLESS an overhang is detected, then 70%.

closed 2024-10-02 by github-actions[bot]

KubaMiszcz, 2025-03-20:
> No. Set min and max to 40% and overhangs to 70%. Always 40% UNLESS an overhang is detected, then 70%.

why is it so hard to add it if in other slicer gives us this option?
set for overhangs, set for small layers, and.... SET FOR THE REST eg 80% not 100%
and min/max layer time is when this time is shorter, what if it takes longer? 100%?