OrcaSlicer/OrcaSlicer #5847: Concentric Shell on Solid Infill not flush with Wall

johndavis48, 2024-06-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

N/A

### Printer

Custom Printer 4mm nozzle

### How to reproduce

Use my STL and select settings from my screenshots.

### Actual results

When using Concentric and 3 wall loops, just within the wall loop on opposing sides there is a gap between wall and bottom shell.
This happened consistently and in exact same place.  I tested using 4 wall loops and the gap was consistently just inside the wall (4th loops).  I have attached pics of the print.  Prints are incomplete because I stopped print after the problem occurred.  

I then tested with No wall loops and there was no gap in the bottom of the print.

John

### Expected results

From my testing, the problem is obviously the slicer creating a gap between the wall and concentric bottom and concentric solid infill.  

### Project file & Debug log uploads

[Table.zip](https://github.com/user-attachments/files/15960667/Table.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Here are pics and screenshots of both 3 wall loop and 4 wall loop.  I purposely raised the z offset so that the wall count is visible.  This is why you will see delamination.  Print was stopped after test was complete.  The material is a bit fuzzy because it is PE with glass fiber.  Let me know if you have any questions because I am eager to move past this.  I am using no walls as a workaround which works for this model but no most of my models.
Thank you,
John

[image]
[image]
[image]

[image]
[image]

johndavis48, 2024-06-24:
If you wonder about  the size  of the model vs the size of the print pics.  My large format printer was larger then Orca max size before version 2.1.0, so I scaled my model to be mm while my printer is sized to cm.  This has worked flawlessly.  Now that Orca has larger bed size available I will change the scale to be mm to mm.

johndavis48, 2024-06-24:
Here is STL
[Table (8).zip](https://github.com/user-attachments/files/15961362/Table.8.zip)

johndavis48, 2024-06-25:
To further test, I created a circle model.  First I printed with Wall loops only.  Then printed the same model with Bottom shell only.  The "Wall only" came out perfect, with exact measurements on every point of the circle.  The "Bottom only" started in an oblong shape and then corrected to a more round shape halfway into the print.  This caused a gap that can be seen in the pic.
(I stopped after 1st layer so prints are a bit flimsy, Sharpie Arrow shows orientation on bed)

This narrows down the problem to Concentric is not slicing properly to match the form of the model, causing gaps and issues flushing up against walls.  As well as other issues.

[image]

[image]

[image]

[image]

closed 2024-06-26 by johndavis48

NickUfer, 2025-05-07:
Why was this closed?

johndavis48, 2025-05-08:
After further testing I determined it was a hardware error so I closed it.