OrcaSlicer/OrcaSlicer #5849: Unable to print on Bambu printers with nozzles sizes other bambu supported nozzles

davidjdixon, 2024-06-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.0

### Operating System (OS)

macOS

### OS Version

14.5

### Printer

Bambu X1 Carbon

### How to reproduce

1. Create a printer profile and configure with a nozzle size larger than 0.8mm (eg, in my case a 1.4mm Revo)
2. Print plate of any model

### Actual results

An error is returned about not matching the nozzle size of the printer, without any option to use the nozzle configuration.

[image]

### Expected results

in version <= 2.0, a warning was produced, but with options to ignore.

[image]

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jjleto, 2024-06-26:
same situation here work partially in 2.1 beta 
[image]

tlhintoq, 2024-07-01:
Windows give me the option to confirm and continue as shown by @jjleto 
2.1.0 latest on 01jun2024

[image]

[image]

davidjdixon, 2024-07-01:
@tlhintoq If your build is from 1st June, then you will be using the 2.1.0 beta (not 2.1.0 as that wasn't released until 21st June), which indicates the related code was either introduced in 2.1.0 RC or 2.1.0.

tlhintoq, 2024-07-02:
@davidjdixon   My bad.  01jul2024  (yesterday)

davidjdixon, 2024-07-26:
@SoftFever is there any chance this recent change can be reverted? This issue prevents using custom nozzle sizes at all with Bambu Lab printers.

Highlander-bdj, 2024-07-30:
In my opinion, #5877 should be listed as a bug not an enchantment.

Anfernee12345, 2024-08-12:
> @SoftFever is there any chance this recent change can be reverted? This issue prevents using custom nozzle sizes at all with Bambu Lab printers.

Still having the same problem.
Cant find any way around it either.

davidjdixon, 2024-08-12:
> > @SoftFever is there any chance this recent change can be reverted? This issue prevents using custom nozzle sizes at all with Bambu Lab printers.
> 
> Still having the same problem. Cant find any way around it either.

Have a try of the latest nightly. While theres been no updates here, it seems that the nightly has reverted this (at least for the moment).

Anfernee12345, 2024-08-12:
> > Still having the same problem. Cant find any way around it either.
> 
> Have a try of the latest nightly. While theres been no updates here, it seems that the nightly has reverted this (at least for the moment).

Yes, thank you.
The latest nightly works, for now at least.

anselor, 2024-08-16:
So, just to clarify for some people. This was a change that Bambu made to block non-standard nozzle diameters. It was patched out in the 2.1 beta but then prior to release another pull from the Bambu code undid the fix so 2.1 was released with this broken. It has subsequently been patched again and is waiting on another release.

davidjdixon, 2024-09-19:
This has been resolved by #5900 and now part of [2.2.0-Beta](https://github.com/SoftFever/OrcaSlicer/releases/tag/v2.2.0-beta)

closed 2024-09-19 by davidjdixon