OrcaSlicer/OrcaSlicer #5850: Move Selected Models To Plate By "Auto Arrange"

Kingsidorak, 2024-06-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Simply a quality of life thing which will probably just be for print farm operators like myself.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

1. Select a plate
2. Select models
3. Click arrange (or use the keybind)
The selected models should then move to the plate, and be arranged.

### Describe alternatives you've considered

manually dragging models to each of the plates

tlhintoq, 2024-07-01:
FYI: The selected plate is where parts land when you either bring them in, or cut/copy then paste.

Does it help your workflow to bring some in, select the next plate, bring the others in, select the next plate.... ?

Or... bring them all in on one (sounds like you're doing).
Select the plate, select the objects to move... cntrl-X, cntrl-V and they should land on the selected plate.
{repeat as needed}

Kingsidorak, 2024-07-01:
My workflow without automation is bring in a ton of parts into the canvas, and dragging them around to different plates depending on what needs to be printed

neutebach, 2024-07-17:
I would like to add to this discussion since I would also like this enhancement.
I provide 3D printing services and often times, orders consist of multiple parts at various quantities. It's most effective to use the auto arrange function to fill plates with instances of each part, for easiest manual separation of the printed parts. I would like to treat each customer order as a project, i.e. load all models into a project and have separate plates for each model.

Currently, all models in the project are arranged to the same (active) plate.
Ideally, selected models should be arranged to the selected/active plate. 

FWIW, this would work easier in Bambu Studio since it seems to auto arrange parts to the active plate **when cloning a model**. Thus, each model can be cloned to a separate plate. Orca however just multiplies the part into the same place, necessating manual arranging afterwards with the above mentioned drawbacks.

Kingsidorak, 2024-07-17:
and this would help reduce us hovering over a model, clicking on it to drag it over, AND THE ENTIRE CANVAS SPINNING BECAUSE YOU CLICKED .00000001 PIXELS AWAY FROM THE MODEL...............

brjohnso1, 2024-08-15:
BUG: 2.2 DEV is currently triggering auto-arrange of all plates and all objects whenever any object on any plate is cloned.

closed 2024-11-21 by github-actions[bot]