OrcaSlicer/OrcaSlicer #5851: Add an override for Z Offset in build plate settings

DarkDoldier, 2024-06-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

If changing build plates, the Z Offset might need to be different because I use a FR4 and a deep pei coated sheet.
Currently, Z Offset is in the printer profile and aims to compensate for bad distance between nozzle and bed. This ignores the fact that the Z Offset often needs to be adapted. I have it Set up for the smooth plate. On the PEI plate I need constantly -0.120 for a good first layer / even adhesion.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

A setting that sets a Klipper call for SET_GCODE_OFFSET Z=-0.120. That should be configurable in Printer Settings.

### Describe alternatives you've considered

I tried it by putting a macro together.

In my orca start code I have:

PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single] Chamber=[chamber_temperature] PRINT_MIN={first_layer_print_min[0]},{first_layer_print_min[1]} PRINT_MAX={first_layer_print_max[0]},{first_layer_print_max[1]} FILAMENT_TYPE="{filament_type[0]}" CURR_BED_TYPE="{curr_bed_type}"

That comes for example in a GCODE file:

PRINT_START EXTRUDER=225 BED=60 Chamber=0 PRINT_MIN=158.603,29.8189 PRINT_MAX=169.889,41.1043 FILAMENT_TYPE="PLA" CURR_BED_TYPE="Textured PEI Plate"

In my PRINT_START macro I have:

{% set CURR_BED_TYPE = params.CURR_BED_TYPE %}

Later on comes:

CONFIGURE_BED_OFFSET CURR_BED_TYPE="{CURR_BED_TYPE}"

And the macro looks like:

[gcode_macro CONFIGURE_BED_OFFSET]
gcode:
    {% set CURR_BED_TYPE = params.CURR_BED_TYPE %}

    {% if CURR_BED_TYPE == ['High Temp Plate'] %}
        SET_GCODE_OFFSET Z=0.0
    {% elif CURR_BED_TYPE == ['Textured PEI Plate'] %}
        SET_GCODE_OFFSET Z=-0.100
    {% else %}
        SET_GCODE_OFFSET Z=0.0
    {% endif %}

But I just can't get it running........... On several tries and changes it did nothing and print without the Z_OFFSET change wich results in having to do it manually and now with the config like it is above I get these error and the print just stops instantly:

Malformed command 'CONFIGURE_BED_OFFSET CURR_BED_TYPE=Textured PEI Plate'
Malformed command 'CONFIGURE_BED_OFFSET CURR_BED_TYPE=Textured PEI Plate'

I just can't get it running.

kylek29, 2024-06-25:
I was going to post this as well. I think the build plate (bed type) system might need to evolve to be like the other systems, I was mapping it out last week. There's so many build plate options now that I have to calibrate for each one because they're all different, and my printer's (BBL X1C) method of calculating it during bed leveling is not that accurate with many.

From looking into the code, it seems to have in-code settings for:

- Build Plate UI Name
- Build Plate Name Alias (displayed on the 3d plate)
- BBL Internal ID (for the plate reader)

Possibly a refinement would to remove these from the code (or set them as the default fallbacks) and put the plate configs into the presets so users can assign the available plates to their printers like they do filaments. The options available could be:

- UI Display Name
- Plate Graphic Override (for changing the title on the 3d view, uses UI Display Name by default)
- ID (for plate scanning)  (this would be the BBL ID or expanded if someone else does plate IDing)
- Logical Name Override (gives the ability to override the displayed UI values so it acts an alias to an existing one, like "Textured G10" aliasing for "PEI Textured", so in G-Code output of curr_bed_type t'd identify as PEI Textured)
- Plate Material (just a helper, so users can put Glass, G10, etc.)
- Plate UI Image Override (to override the inherited bed graphic)
- Filament Type Warning List (list of filaments that if you try to print with it and this plate, it pops a warning)

##### Offsets & Modifiers
- Z-Offset value
- Z-Offset Overrides Machine Z-Offset (if true, it would replace the Z-Offset in the Machine profile, else it acts a modifier --> e.g. Machine Z = -0.04, plate = 0.02, output would be -0.02).
- Use Filament Temp Plate Setting (select from the available temp setting boxes e.g. cool, engineering, hot, textured)
- Modify Temp Setting (+/- the value) First Layer (adds/subtracts a value to the first layer filament temp) (purpose is that for some specific plates you may want to run a tad hotter/colder)
- Modify Temp Setting All Other Layers (same as above, for all layers)
- Add Additional Temp Soak Wait Time (once it reaches the bed-temp target, it waits X seconds to soak before proceeding on) (good for thicker plates, since the thermal couples are usually on the underside with the heating element).

DarkDoldier, 2024-07-07:
I was able to get some of the things above to work via macro. I uploaded it as a repository.

closed 2024-10-14 by github-actions[bot]