OrcaSlicer/OrcaSlicer #5855: Model sketches have disappeared in Gcode since version 2.1( BIGTREETECH tft v3.1)

Rsoft1976, 2024-06-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

Artillery SX1

### How to reproduce

starting from version 2.1 the picture disappeared in the settings, I also can’t select a type

there were settings
2.0
[image]

2.1 
[image]

### Actual results

are added at the beginning of the line that does not work with Gcode
[image]

### Expected results

remove these lines at the very beginning
[image]

### Project file & Debug log uploads

no

### Checklist of files to include

- [ ] Log file
- [ ] Project file

blackadar, 2024-06-26:
Also missing thumbnails with the latest release

blackadar, 2024-07-03:
Ended up resolving this by modifying my printer config in Orca, it was missing the thumbnail size option. Once I put that back thumbnails started generating again.

Rsoft1976, 2024-07-05:
what screen model?

blasksun, 2024-08-05:
artillery genius pro, same problem after updating osca to 2.1 or latest GitHub thumbnails wont show up on the printer

Rsoft1976, 2024-11-03:
Dear developer, will you fix this problem or not?

PR #7488 (2024-11-30): don't create HEADER_BLOCK_START if using BTT_TFT thumbnails
For some time now, the generation of thumbnails when using the BTT_TFT format is not working because the first line of the G-code does not start with the generated thumbnail.

So I tried to bypass the generation of the **`HEADER_BLOCK_START`** using an if statement.

But unfortunately I am not skilled enough to do this myself because it does not compile.

Maybe someone can help me with this. 😊

@Noisyfox
@igiannakas
@cochcoder

# @igiannakas provided the code for this PR that made it work! 👍

This should fix #5855

Thanks in advance
closed 2024-11-30 by SoftFever